sustainable-computing-io/kepler #333: Document deployment settings and cli options

rootfs, 2022-10-24:


jichenjc, 2022-10-31:
is this something should be tracked in sustainable computing overall doc issue? 
e.g we need update here https://sustainable-computing.io/html/installation/index.html ?

rootfs, 2022-10-31:
@jichenjc yes, eventually all the doc will go to the https://github.com/sustainable-computing-io/kepler-doc repo

husky-parul, 2022-11-09:
@jichenjc 
This is our documentation repo https://github.com/sustainable-computing-io/kepler-doc
When you are done with it, please create a PR on the main branch and add me as the reviewer. We use sphinx to build the html so the content goes inside [docs/source](https://github.com/sustainable-computing-io/kepler-doc/tree/main/docs/source/)

rootfs, 2022-12-07:
bump up this issue

rootfs, 2022-12-07:
- [x] deployment scenarios with model server, estimator, and kepler: This is supported in the Operator instruction [here](https://github.com/sustainable-computing-io/kepler-doc/blob/main/docs/usage/general_config.md)
- [ ] deployment scenarios and cli options recommended for VM and baremetal: This is also documented in the Kepler installation [here](https://github.com/sustainable-computing-io/kepler-doc/blob/main/docs/installation/kepler.md)
- [ ] troubleshooting guide.

jichenjc, 2022-12-08:
I haven't try https://github.com/sustainable-computing-io/kepler/issues/448 personally 
but I think it might impact 
`deployment scenarios and cli options recommended for VM and baremetal: This is also documented in the Kepler installation` ？

sunya-ch, 2022-12-08:
Is there anyone already assigned to update the doc? Otherwise, I will create a PR for that.

rootfs, 2022-12-08:
trouble shooting guide PR is here https://github.com/sustainable-computing-io/kepler-doc/pull/34

closed 2022-12-09 by rootfs