sustainable-computing-io/kepler #334: Document metrics and their usage

rootfs, 2022-10-24:


jichenjc, 2022-10-25:
I knew @marceloamaral is working on (already done?) container level metrics
should we first record old(legacy) metrics then continue improve?

marceloamaral, 2022-10-25:
This is a good idea, I will work on it! I think we should already document the new container-level metrics....
Btw, we have some good discussion of how to use the metrics in #291

husky-parul, 2022-11-09:
Hi @jichenjc and @marceloamaral 
This is our documentation repo https://github.com/sustainable-computing-io/kepler-doc
When you are done with it, please create a PR on the main branch and add me as the reviewer. We use sphinx to build the html so the content goes inside [docs/source](https://github.com/sustainable-computing-io/kepler-doc/tree/main/docs/source/)

We are going to use this issue to track the PRs so please link this issue in your PR.
Thanks.

marceloamaral, 2022-11-11:
@husky-parul I created a PR with the detailed description of the metrics.
https://github.com/sustainable-computing-io/kepler-doc/pull/25

jichenjc, 2022-12-08:
I think we covered this already

closed 2022-12-08 by jichenjc