sustainable-computing-io/kepler #348: how the upstream build process different to local build?

jichenjc, 2022-11-01:
**Describe the bug**
A clear and concise description of what the bug is.

using my local build image through `make`
```
# kubectl logs kepler-exporter-nhc7c -n kepler
I1101 07:30:48.662096       1 exporter.go:74] Kepler running on version: v0.3-71-g1b92289
```

using upstream latest 
```
# kubectl logs kepler-exporter-zmf99 -n kepler
I1101 07:51:14.034145       1 exporter.go:74] Kepler running on version:
```

@SamYuan1990 what's the latest image build ? something about the process might be helpful here

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

marceloamaral, 2022-11-01:
The image update is made in [workflows/image.yml](https://github.com/sustainable-computing-io/kepler/blob/main/.github/workflows/image.yml). A new image will be created every time that the `main` is updated.

The [_build_local](https://github.com/sustainable-computing-io/kepler/blob/main/Makefile#L67) and  [_build_containerized](https://github.com/sustainable-computing-io/kepler/blob/main/Makefile#L84) are using different parameters. This might be the problem...

jichenjc, 2022-11-01:
I am wondering https://github.com/sustainable-computing-io/kepler/blob/main/.github/workflows/image.yml#L20
is the root cause ..

jichenjc, 2022-11-01:
ok, saw your comment, actually with make we can see
```
Step 13/21 : RUN make clean $MAKE_TARGET SOURCE_GIT_TAG=$SOURCE_GIT_TAG BIN_TIMESTAMP=$BIN_TIMESTAMP
 ---> Running in 8693e36449b7
rm -f -r '_output/bin'

```

the target is cross-build-linux-amd64 and this build will call _build_local anyway.. so I think it's still same process for build the kepler execute file

marceloamaral, 2022-11-01:
@jichenjc. Just `make` actually run: `make _build_containerized ARCH=amd64`

[_build_containerized](https://github.com/sustainable-computing-io/kepler/blob/main/Makefile#L84) has the following arguments:
```Makefile
OS := $(shell go env GOOS)
ARCH := $(shell go env GOARCH)
SOURCE_GIT_TAG=$(SOURCE_GIT_TAG)
BIN_TIMESTAMP=$(BIN_TIMESTAMP)
ARCH=$(ARCH) 
MAKE_TARGET="cross-build-linux-$(ARCH)"
```

[_build_local](https://github.com/sustainable-computing-io/kepler/blob/main/Makefile#L67) has the following arguments:
```Makefile
GOOS=$(GOOS) 
GOARCH=$(GOARCH)
${GO_BUILD_TAGS}
$(LDFLAGS)
```

The variable that is adding the version is the `LDFLAGS`, but the `_build_containerized` does not have this variable. You probably need to add it.

The Makefile has the variable `GO_LD_FLAGS := $(GC_FLAGS) -ldflags "-X $(LD_FLAGS)" $(CFLAGS)` that is not being used anywhere. 
Probably both the `_build_containerized` and `_build_local` should use the `GO_LD_FLAGS`.

By the way, `_build_containerized` should also have `GO_BUILD_TAGS`.

jichenjc, 2022-11-01:
thanks for the detailed info, however, curious why my local test build `make` can create container with LDFLAGS...
something need dig into :) 

will submit a PR to this

marceloamaral, 2022-11-01:
Humm, you are right @jichenjc ...

I'm using and the [_build_containerized](https://github.com/sustainable-computing-io/kepler/blob/main/Makefile#L84) and it has the version: `v0.3-82-ge5113f5-dirty` in the logs.

So, the GO_LD_FLAGS is being computed and used when building the image within the container....

The problem might be something else...

closed 2022-11-22 by SamYuan1990