sustainable-computing-io/kepler #350: Improve integration test suite to test basic metric query

sallyom, 2022-11-01:
can we reuse [#162?](https://github.com/sustainable-computing-io/kepler/issues/162)

rootfs, 2022-11-01:
@sallyom sounds good, please list the previously opened issues for this task.

sallyom, 2022-11-01:
previous work:
- https://github.com/sustainable-computing-io/kepler/pull/283
- https://github.com/sustainable-computing-io/kepler/pull/292

sallyom, 2022-11-01:
From #162 @rootfs & @SamYuan1990 

> The service readiness detection is to be supported in kepler-operator

> > so what's our scope of integration test?

Will open a design doc to discuss integration test goals

SamYuan1990, 2022-11-07:
is there any thing we left in this issue?

sallyom, 2022-11-07:
@SamYuan1990 @rootfs this merged last week, https://github.com/sustainable-computing-io/kepler/pull/357  so I believe this issue is completed! :partying_face: :tada:

sallyom, 2022-11-07:
https://github.com/sustainable-computing-io/kepler/issues/349 will add more in-depth integration tests. The current suite satisfies this issue.

closed 2022-11-07 by sallyom