sustainable-computing-io/kepler #352: VM: all node / pod energy report 0 (again..)

jichenjc, 2022-11-02:
**Describe the bug**
A clear and concise description of what the bug is.

after use latest update with a few enhancement, my pod/node report energy become 0 again.. 
switch to v0.3 I can see the data correctly reported ,...

latest
```
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0kepler_node_energy_stat{cpu_architecture="Haswell",node_block_devices_used="0",node_curr_bytes_read="0",node_curr_bytes_writes="0",node_curr_cache_miss="0",node_curr_container_cpu_usage_seconds_total="0",node_curr_container_memory_working_set_bytes="0",node_curr_cpu_cycles="0",node_curr_cpu_instr="0",node_curr_cpu_time="0",node_curr_energy_in_core_joule="0",node_curr_energy_in_dram_joule="0",node_curr_energy_in_gpu_joule="0",node_curr_energy_in_other_joule="0",node_curr_energy_in_pkg_joule="0",node_curr_energy_in_uncore_joule="0",node_name="jitest40"} 0

```
v0.3
```
# TYPE node_curr_energy_in_core_joule gauge
node_curr_energy_in_core_joule{instance="jitest40"} 0.026
# HELP node_curr_energy_in_dram_joule node_ current energy consumption in dram (joule)
# TYPE node_curr_energy_in_dram_joule gauge
node_curr_energy_in_dram_joule{instance="jitest40"} 0.708
# HELP node_curr_energy_in_gpu_joule node_ current energy consumption in gpu (joule)
# TYPE node_curr_energy_in_gpu_joule gauge
node_curr_energy_in_gpu_joule{instance="jitest40"} 0
# HELP node_curr_energy_in_other_joule node_ current energy consumption in other (joule)
# TYPE node_curr_energy_in_other_joule gauge
node_curr_energy_in_other_joule{instance="jitest40"} 0
# HELP node_curr_energy_in_pkg_joule node_ current energy consumption in pkg (joule)
# TYPE node_curr_energy_in_pkg_joule gauge
node_curr_energy_in_pkg_joule{instance="jitest40"} 0.026
# HELP node_curr_energy_in_uncore_joule node_ current energy consumption in uncore (joule)
# TYPE node_curr_energy_in_uncore_joule gauge
node_curr_energy_in_uncore_joule{instance="jitest40"} 0
# HELP node_curr_energy_joule node_ current energy consumption (joule)
# TYPE node_curr_energy_joule gauge
node_curr_energy_joule{instance="jitest40"} 0.026

```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

marceloamaral, 2022-11-02:
What about the other metrics?
`kepler_node_core_joules_total`
`kepler_node_package_joules_total`

marceloamaral, 2022-11-02:
I can see the values in my system
```
# TYPE kepler_node_dram_joules_total counter
kepler_node_dram_joules_total{instance="kind-control-plane",package="0",source="rapl"} 13151.144
kepler_node_dram_joules_total{instance="kind-control-plane",package="1",source="rapl"} 49141.363
kepler_node_dram_joules_total{instance="kind-control-plane",package="2",source="rapl"} 45732.573
kepler_node_dram_joules_total{instance="kind-control-plane",package="3",source="rapl"} 53145.553
# HELP kepler_node_nodeInfo Labeled node information
# TYPE kepler_node_nodeInfo counter
kepler_node_nodeInfo{cpu_architecture="Cascade Lake"} 1
# HELP kepler_node_package_joules_total Aggregated RAPL value in package (socket) in joules
# TYPE kepler_node_package_joules_total counter
kepler_node_package_joules_total{instance="kind-control-plane",package="0",source="rapl"} 1930.436
kepler_node_package_joules_total{instance="kind-control-plane",package="1",source="rapl"} 105215.206
kepler_node_package_joules_total{instance="kind-control-plane",package="2",source="rapl"} 171142.694
kepler_node_package_joules_total{instance="kind-control-plane",package="3",source="rapl"} 188908.955
```

But the energy stat is empty:
```
# HELP kepler_node_energy_stat Several labeled node metrics
# TYPE kepler_node_energy_stat counter
kepler_node_energy_stat{cpu_architecture="Cascade Lake",node_block_devices_used="0",node_curr_bytes_read="0",node_curr_bytes_writes="0",node_curr_cache_miss="0",node_curr_container_cpu_usage_seconds_total="0",node_curr_container_memory_working_set_bytes="0",node_curr_cpu_cycles="0",node_curr_cpu_instr="0",node_curr_cpu_time="0",node_curr_energy_in_core_joule="0",node_curr_energy_in_dram_joule="0",node_curr_energy_in_gpu_joule="0",node_curr_energy_in_other_joule="0",node_curr_energy_in_pkg_joule="0",node_curr_energy_in_uncore_joule="0",node_name="kind-control-plane"} 0
```

I will investigate that next week (after the holiday in Japan)

----
Btw, the `kepler_node_package_joules_total` use the `p.NodeMetrics.EnergyInPlatform.Agg()`
And the `kepler_node_energy_stat` use the  `p.NodeMetrics.EnergyInPlatform.Curr()`
So it might be something related to the `Agg()` and `Curr()` functions.

jichenjc, 2022-11-03:
```
root@jitest41:~/kepler# curl localhost:9102/metrics | grep kepler_node_core_joules_total
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 66669    0 66669    0     0  21.1M      0 --:--:-- --:--:-- --:--:-- 21.1M
root@jitest41:~/kepler# curl localhost:9102/metrics | grep kepler_node_energy_stat
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
# HELP kepler_node_energy_stat Several labeled node metrics
1# TYPE kepler_node_energy_stat counter
0kepler_node_energy_stat{cpu_architecture="Haswell",node_block_devices_used="0",node_curr_bytes_read="0",node_curr_bytes_writes="0",node_curr_cache_miss="0",node_curr_container_cpu_usage_seconds_total="0",node_curr_container_memory_working_set_bytes="0",node_curr_cpu_cycles="0",node_curr_cpu_instr="0",node_curr_cpu_time="0",node_curr_energy_in_core_joule="0",node_curr_energy_in_dram_joule="0",node_curr_energy_in_gpu_joule="0",node_curr_energy_in_other_joule="0",node_curr_energy_in_pkg_joule="0",node_curr_energy_in_uncore_joule="0",node_name="jitest41"} 0
0 66675    0 66675    0     0  21.1M      0 --:--:-- --:--:-- --:--:-- 21.1M
root@jitest41:~/kepler#

```

didn't see this package info in output, need check why ..

jichenjc, 2022-11-03:
all pod energy 0 as well ..

```
16",total_cpu_cycles="0",total_cpu_instr="0",total_cpu_time="76764"} 0
kepler_pod_energy_stat{block_devices_used="0",command="prometheus",container_name="config-reloader",curr_bytes_read="0",curr_bytes_writes="0",curr_cache_miss="0",curr_container_cpu_usage_seconds_total="0",curr_container_memory_working_set_bytes="0",curr_cpu_cycles="0",curr_cpu_instr="0",curr_cpu_time="0",pod_name="alertmanager-main-0",pod_namespace="monitoring",total_bytes_read="0",total_bytes_writes="0",total_cache_miss="0",total_container_cpu_usage_seconds_total="0",total_container_memory_working_set_bytes="11399168",total_cpu_cycles="0",total_cpu_instr="0",total_cpu_time="41945"} 0
kepler_pod_energy_stat{block_devices_used="0",command="prometheus",container_name="config-reloader",curr_bytes_read="0",curr_bytes_writes="0",curr_cache_miss="0",curr_container_cpu_usage_seconds_total="0",curr_container_memory_working_set_bytes="0",curr_cpu_cycles="0",curr_cpu_instr="0",curr_cpu_time="0",pod_name="alertmanager-main-1",pod_namespace="monitoring",total_bytes_read="0",total_bytes_writes="0",total_cache_miss="0",total_container_cpu_usage_seconds_total="0",total_container_memory_working_set_bytes="12943360",total_cpu_cycles="0",total_cpu_instr="0",total_cpu_time="21018"} 0
kepler_pod_energy_stat{block_devices_used="0",command="prometheus",container_name="config-reloader",curr_bytes_read="0",curr_bytes_writes="0",curr_cache_miss="0",curr_container_cpu_usage_seconds_total="0",curr_container_memory_working_set_bytes="0",curr_cpu_cycles="0",curr_cpu_instr="0",curr_cpu_time="0",pod_name="alertmanager-main-2",pod_namespace="monitoring",total_bytes_read="0",total_bytes_writes="0",total_cache_miss="0",total_container_cpu_usage_seconds_total="0",total_container_memory_working_set_bytes="10657792",total_cpu_cycles="0",total_cpu_instr="0",total_cpu_time="47091"} 0
kepler_pod_energy_stat{block_devices_used="0",command="prometheus",container_name="config-reloader",curr_bytes_read="0",curr_bytes_writes="0",curr_cache_miss="0",curr_container_cpu_usage_seconds_total="0",curr_container_memory_working_set_bytes="0",curr_cpu_cycles="0",curr_cpu_instr="0",curr_cpu_time="0",pod_name="prometheus-k8s-0",pod_namespace="monitoring",total_bytes_read="0",total_bytes_writes="0",total_cache_miss="0",total_container_cpu_usage_seconds_total="0",total_container_memory_working_set_bytes="13180928",total_cpu_cycles="0",total_cpu_instr="0",total_cpu_time="39257"} 0

```

will dig more on this

rootfs, 2022-11-03:
The VM estimator is moved, can you enable estimator? https://github.com/sustainable-computing-io/kepler-estimator#patch-estimator-sidecar-to-kepler-daemonset
cc @sunya-ch

jichenjc, 2022-11-04:
yes, I enabled it (some doc issue , will open other issues to track in that repo), thanks for the tip

jichenjc, 2022-11-05:
I doubt current code works for Ubuntu at all

I got 2 env , both Ubuntu , one with cgroupv2, one not 

```
root@jitest40:/sys/fs# cat /sys/fs/cgroup/cgroup.controllers
cat: /sys/fs/cgroup/cgroup.controllers: No such file or directory

root@jitest41:/sys/fs# cat /sys/fs/cgroup/cgroup.controllers
cpuset cpu io memory pids rdma

```

both looks no scope at all 
```
# find ./ -name '*.scope'
./cgroup/memory/user.slice/user-1000.slice/session-11.scope
./cgroup/memory/init.scope
./cgroup/blkio/init.scope
./cgroup/devices/init.scope
./cgroup/pids/user.slice/user-1000.slice/session-11.scope
./cgroup/pids/init.scope
./cgroup/cpu,cpuacct/init.scope
./cgroup/systemd/user.slice/user-1000.slice/user@1000.service/init.scope
./cgroup/systemd/user.slice/user-1000.slice/session-11.scope
./cgroup/systemd/init.scope
./cgroup/unified/user.slice/user-1000.slice/user@1000.service/init.scope
./cgroup/unified/user.slice/user-1000.slice/session-11.scope
./cgroup/unified/init.scope

root@jitest41:/sys/fs# find ./ -name '*.scope'
./cgroup/user.slice/user-1000.slice/user@1000.service/init.scope
./cgroup/user.slice/user-1000.slice/session-1.scope
./cgroup/init.scope

```

but https://github.com/sustainable-computing-io/kepler/blob/main/pkg/cgroup/slice_handler.go#L155 expecting a scope suffix file or the compare failed totally and no data reported 

and I am curious on our current design @marceloamaral 

no cgropu v2 env
```
root@jitest40:/sys/fs# ls ./cgroup/memory/kubepods.slice/kubepods-besteffort.slice/kubepods-besteffort-podfe2bf65f_739a_4296_a14b_1dd455bbd48b.slice/
cgroup.clone_children       memory.kmem.max_usage_in_bytes      memory.limit_in_bytes            memory.stat
cgroup.event_control        memory.kmem.slabinfo                memory.max_usage_in_bytes        memory.swappiness
cgroup.procs                memory.kmem.tcp.failcnt             memory.move_charge_at_immigrate  memory.usage_in_bytes
memory.failcnt              memory.kmem.tcp.limit_in_bytes      memory.numa_stat                 memory.use_hierarchy
memory.force_empty          memory.kmem.tcp.max_usage_in_bytes  memory.oom_control               notify_on_release
memory.kmem.failcnt         memory.kmem.tcp.usage_in_bytes      memory.pressure_level            tasks
memory.kmem.limit_in_bytes  memory.kmem.usage_in_bytes          memory.soft_limit_in_bytes

root@jitest40:/sys/fs# ls cgroup/cpu/kubepods.slice/kubepods-besteffort.slice/kubepods-besteffort-podfe2bf65f_739a_4296_a14b_1dd455bbd48b.slice/
cgroup.clone_children  cpuacct.usage         cpuacct.usage_percpu_sys   cpuacct.usage_user  cpu.shares      cpu.uclamp.min
cgroup.procs           cpuacct.usage_all     cpuacct.usage_percpu_user  cpu.cfs_period_us   cpu.stat        notify_on_release
cpuacct.stat           cpuacct.usage_percpu  cpuacct.usage_sys          cpu.cfs_quota_us    cpu.uclamp.max  tasks

```

with cgroup v2 

```
root@jitest41:/sys/fs# ls ./cgroup/kubepods.slice/kubepods-burstable.slice/kubepods-burstable-podee30fe39_4dd5_48e6_b101_13cadeff45bb.slice/
cgroup.controllers      cgroup.subtree_control  cpuset.cpus.partition  cpu.weight.nice  memory.events.local  memory.stat
cgroup.events           cgroup.threads          cpuset.mems            io.max           memory.high          pids.current
cgroup.freeze           cgroup.type             cpuset.mems.effective  io.pressure      memory.low           pids.events
cgroup.max.depth        cpu.max                 cpu.stat               io.stat          memory.max           pids.max
cgroup.max.descendants  cpu.pressure            cpu.uclamp.max         io.weight        memory.min           rdma.current
cgroup.procs            cpuset.cpus             cpu.uclamp.min         memory.current   memory.oom.group     rdma.max
cgroup.stat             cpuset.cpus.effective   cpu.weight             memory.events    memory.pressure
```

should we abstract different distro?

to me ,seems existing code expect we should have those files but cgroupv1 have them (no io.stat)
but cgroupv2 doesn't? 
https://github.com/sustainable-computing-io/kepler/blob/main/pkg/cgroup/stat_reader.go#L25

jichenjc, 2022-11-05:
I know this used to work on my env before so I am not sure how the refactory recently impact those logic..
can you help provide some insight on where to find 
https://github.com/sustainable-computing-io/kepler/blob/main/pkg/cgroup/stat_reader.go#L25
https://github.com/sustainable-computing-io/kepler/blob/main/pkg/cgroup/stat_reader.go#L32
https://github.com/sustainable-computing-io/kepler/blob/main/pkg/cgroup/stat_reader.go#L39

under which condition should I am able to use them?

marceloamaral, 2022-11-07:
Hum, so the problem is with the cgroup metrics?

jichenjc, 2022-11-07:
yes, the cgroup metrics is not able to be found , all report 0 might be related to this cgroup metric is empty..
can't find scope file ,then no available metrics and not get all those cgroup metcis? 
I am not sure the design logic (need more read) but seems without cgroup metrics there is no way to report container energy consumption? 

```
I1107 03:16:32.235670       1 utils.go:43] Available cgroup metrics from cgroup: []

```

marceloamaral, 2022-11-07:
No, we can report the container energy consumption without the cgroup metric...
The most important metrics are the hardware counters from BPF.

----

And yes, it seems that we need to improve the cgroup metric collection logic

jichenjc, 2022-11-07:
>>>The most important metrics are the hardware counters from BPF.

ok, so my env didn't have BPF functional (fully) , and cgroup metics not working well as well
that's the reason I got 0 ... so first should improve BPF then optimize cgroup as well...

marceloamaral, 2022-11-07:
Yes, I added a TODO in the code to improve this...
https://github.com/sustainable-computing-io/kepler/blob/bf17c6bfa1062e4d557d0aeb1ce2ffc67ade803e/pkg/collector/container_energy_collector.go#L33
I will create a new issue for this

marceloamaral, 2022-11-07:
#364

jichenjc, 2022-11-10:
I am seeing this 

```
I1110 09:21:38.513914       1 slice_handler.go:165] Not able to find any valid .scope file in /sys/fs/cgroup/kubepods.slice, this likely cause all cgroup metrics to be 0
I1110 09:21:38.530883       1 bcc_attacher.go:143] hardeware counter metr true
I1110 09:21:38.530930       1 utils.go:41] Available counter metrics: [cache_miss cpu_cycles cpu_instr]
I1110 09:21:38.530945       1 utils.go:42] Available cgroup metrics from cgroup: []

```

but I am seeing 
```
# find /sys/fs/cgroup/kubepods.slice/ -name '*.scope'
/sys/fs/cgroup/kubepods.slice/kubepods-burstable.slice/kubepods-burstable-pod083a9954_35b5_4ad4_a9f9_2da4a7b84f68.slice/crio-0404df9d50ccbc522ceb3d7c01889fdea11480c4e09669119f4dc005e3ae7aca.scope
/sys/fs/cgroup/kubepods.slice/kubepods-burstable.slice/kubepods-burstable-pod083a9954_35b5_4ad4_a9f9_2da4a7b84f68.slice/crio-513537c9a2e7088e0cd6c39b45fa45869b9bf76c0302035da36af1a6470a84a6.scope
/sys/fs/cgroup/kubepods.slice/kubepods-burstable.slice/kubepods-burstable-pod65873b62bb813af6737b014e65c9bd35.slice/crio-b6f7df46bbbbc9ff460fa0f4dab6786b8b661b757f29f88719a5a7cd91f9cb3d.scope
/sys/fs/cgroup/kubepods.slice/kubepods-burstable.slice/kubepods-burstable-pod65873b62bb813af6737b014e65c9bd35.slice/crio-a0b8cfad2647368b33ca9b8e06f247a641c97d50c662129235534dfc457b2eec.scope
/sys/fs/cgroup/kubepods.slice/kubepods-burstable.slice/kubepods-burstable-podd2429e27868c81d07238891b6b7be89c.slice/crio-86f7637b1828a19bfd5f7b159daa945224b58b434913685fa590ee4ad0b36bed.scope
/sys/fs/cgroup/kubepods.slice/kubepods-burstable.slice/kubepods-burstable-podd2429e27868c81d07238891b6b7be89c.slice/crio-834a916ffe2f4cfed0e20743f44e27a9e7b5bdc56c28a780fd5f91091eb4c3d2.scope
/sys/fs/cgroup/kubepods.slice/kubepods-burstable.slice/kubepods-burstable-pod33a7e72a953ba5450e0b613cb4f8de58.slice/crio-51dad52387b29b69cdfb5b75b1675695bce532c7589c698da4adf0116a446e5b.scope
/sys/fs/cgroup/kubepods.slice/kubepods-burstable.slice/kubepods-burstable-pod33a7e72a953ba5450e0b613cb4f8de58.slice/crio-46c131c26438b0603014ce05d932cf066832fa2ddcb01fbb58600dac9bc0ffe8.scope
/sys/fs/cgroup/kubepods.slice/kubepods-burstable.slice/kubepods-burstable-podaa21f62a_9600_48f4_b71d_0d26ba09beb7.slice/crio-07f4073af123640c64907d2dbdf006db8d3dfd0bda221544f6ac14c6d8ecc22d.scope
/sys/fs/cgroup/kubepods.slice/kubepods-burstable.slice/kubepods-burstable-podaa21f62a_9600_48f4_b71d_0d26ba09beb7.slice/crio-ccd551e7c6ed0a650a527e376bc5851701d26dda86e3140d6c6e73ed625790bc.scope
/sys/fs/cgroup/kubepods.slice/kubepods-burstable.slice/kubepods-burstable-podaa21f62a_9600_48f4_b71d_0d26ba09beb7.slice/crio-2e1382b2e7962490f3dfcd696166ec94aa927a8e55875ea77a11b468f4fff530.scope
/sys/fs/cgroup/kubepods.slice/kubepods-burstable.slice/kubepods-burstable-pod4c2045d2a09e6d8ec3d94ec6e7bad26d.slice/crio-8dd7856134ea5611fe1ca3de11cf1c0ed585b466bd83c66bd2879fa2edd4e870.scope
/sys/fs/cgroup/kubepods.slice/kubepods-burstable.slice/kubepods-burstable-pod4c2045d2a09e6d8ec3d94ec6e7bad26d.slice/crio-2faa6f2deccf308486b9fcdad9fc9c4e8d318660d64fcaebaa76d7b7db0d367e.scope
/sys/fs/cgroup/kubepods.slice/kubepods-burstable.slice/kubepods-burstable-pod0ee202c1_f0ab_49f8_a615_5f7ce17812b0.slice/crio-f050043ea55c2c091de475ed7085dc94840c180f6d12d1975644025340fe38c9.scope
/sys/fs/cgroup/kubepods.slice/kubepods-burstable.slice/kubepods-burstable-pod0ee202c1_f0ab_49f8_a615_5f7ce17812b0.slice/crio-326a528385485dfa8c3e6507eab08dfda9bd9eeedf251cbe19b8c16ee5dd823c.scope
/sys/fs/cgroup/kubepods.slice/kubepods-besteffort.slice/kubepods-besteffort-podd4f512ba_d18f_41ef_a3a7_0d169174b9ff.slice/crio-8d3f22757b48b14b1b1e3c01e2888fe07c49733b6c75a8c3a11026cec7b8875e.scope
/sys/fs/cgroup/kubepods.slice/kubepods-besteffort.slice/kubepods-besteffort-podd4f512ba_d18f_41ef_a3a7_0d169174b9ff.slice/crio-12b270d3b4ec20ad1fe09889bbf7faa153309d691adbaa660b110289a41f6434.scope
```

I think current logic for checking scope might have some problem... @marceloamaral

PR #377 (2022-11-11): fix: search scope file update for not found issue
fix #352
closed 2022-11-11 by rootfs