sustainable-computing-io/kepler #354: Improve unit test coverage to 50+%

SamYuan1990, 2022-11-02:


SamYuan1990, 2022-11-18:
maybe we need to clarify #411

SamYuan1990, 2022-11-29:
ref to https://github.com/sustainable-computing-io/kepler/actions/runs/3574669171/jobs/6010219583

SamYuan1990, 2022-11-29:
coverage: 0.0% of statements
ok  	github.com/sustainable-computing-io/kepler/pkg/bpfassets	0.068s
coverage: 0.0% of statements
ok  	github.com/sustainable-computing-io/kepler/pkg/bpfassets/attacher	0.059s
coverage: 51.1% of statements
ok  	github.com/sustainable-computing-io/kepler/pkg/cgroup	0.116s
coverage: 39.0% of statements
ok  	github.com/sustainable-computing-io/kepler/pkg/collector	0.096s
coverage: 51.0% of statements
ok  	github.com/sustainable-computing-io/kepler/pkg/collector/metric	0.079s
coverage: 60.0% of statements
ok  	github.com/sustainable-computing-io/kepler/pkg/config	0.052s
coverage: 9.3% of statements
ok  	github.com/sustainable-computing-io/kepler/pkg/kubelet	0.066s
coverage: 0.0% of statements
ok  	github.com/sustainable-computing-io/kepler/pkg/manager	0.064s
coverage: 55.4% of statements
ok  	github.com/sustainable-computing-io/kepler/pkg/model	0.272s
coverage: 0.0% of statements
ok  	github.com/sustainable-computing-io/kepler/pkg/model/estimator/local	0.062s
coverage: 0.0% of statements
ok  	github.com/sustainable-computing-io/kepler/pkg/model/estimator/sidecar	0.043s
?   	github.com/sustainable-computing-io/kepler/pkg/model/types	[no test files]
coverage: 58.6% of statements
ok  	github.com/sustainable-computing-io/kepler/pkg/utils	0.054s
coverage: [no statements]
ok  	github.com/sustainable-computing-io/kepler/pkg/version	0.064s

SamYuan1990, 2022-11-29:
- [ ]  github.com/sustainable-computing-io/kepler/pkg/bpfassets
- [ ] github.com/sustainable-computing-io/kepler/pkg/bpfassets/attacher
- [ ] github.com/sustainable-computing-io/kepler/pkg/collector
- [ ] github.com/sustainable-computing-io/kepler/pkg/kubelet
- [ ] github.com/sustainable-computing-io/kepler/pkg/manager 
- [ ] github.com/sustainable-computing-io/kepler/pkg/model/estimator/local 
- [ ] github.com/sustainable-computing-io/kepler/pkg/model/estimator/sidecar
- [ ] github.com/sustainable-computing-io/kepler/pkg/version

rootfs, 2022-12-07:
the coverage reached 51%

closed 2022-12-07 by rootfs