sustainable-computing-io/kepler #355: container_metric.go:176] cannot extract: block_devices_used

jichenjc, 2022-11-03:
**Describe the bug**
A clear and concise description of what the bug is.

with VM and --v=5 

see in logs, not sure what happend?
```
        kubelets: map[container_cpu_usage_seconds_total:0 (7) container_memory_working_set_bytes:0 (12730368)]

I1103 07:36:37.880281       1 metric_collector.go:129] {map[cpu_architecture:158629] 0 (0) 0 (0) 0 (0) 0 (0) 0 (0) 0 (0) 0 (0)}
I1103 07:36:37.880296       1 metric_collector.go:131] Collector Update elapsed time: %s 298.653777ms
I1103 07:36:38.408235       1 container_metric.go:176] cannot extract: block_devices_used
I1103 07:36:38.408321       1 container_metric.go:176] cannot extract: block_devices_used
I1103 07:36:38.408353       1 container_metric.go:176] cannot extract: block_devices_used
I1103 07:36:38.408387       1 container_metric.go:176] cannot extract: block_devices_used
I1103 07:36:38.408419       1 container_metric.go:176] cannot extract: block_devices_used
I1103 07:36:38.408449       1 container_metric.go:176] cannot extract: block_devices_used
I1103 07:36:38.408476       1 container_metric.go:176] cannot extract: block_devices_used
I1103 07:36:38.408514       1 container_metric.go:176] cannot extract: block_devices_used

```

rootfs, 2022-11-03:
This might require turning on block io accounting, [this](https://man.archlinux.org/man/systemd.resource-control.5.en) is the instruction

jichenjc, 2022-11-04:
ok, so should we consider cgroup v1 support or stick to v2 ? 
if it's v2 only, I might need switch my env or it's v1+v2, may need a PR to fix if possible ...

rootfs, 2022-11-04:
@jichenjc please see if cgroup v2 works first.

jichenjc, 2022-11-17:
already cgroupv2 now
still seeing this 
I1117 00:44:32.693941       1 container_metric.go:194] cannot extract: block_devices_used

the metrics are: 
```
I1117 00:44:27.021157       1 utils.go:38] Available counter metrics: [cache_miss cpu_cycles cpu_instr]
I1117 00:44:27.021178       1 utils.go:39] Available cgroup metrics from cgroup: [cgroupfs_ioread_bytes cgroupfs_iowrite_bytes cgroupfs_memory_usage_bytes cgroupfs_cpu_usage_us cgroupfs_system_cpu_usage_us cgroupfs_user_cpu_usage_us]
I1117 00:44:27.021186       1 utils.go:40] Available cgroup metrics from kubelet: [container_cpu_usage_seconds_total container_memory_working_set_bytes]
I1117 00:44:27.021189       1 utils.go:41] Available I/O metrics: [bytes_read bytes_writes]
```

seems config.BlockDevicesIO =  "block_devices_used" is not used at all?

rootfs, 2022-11-17:
I think if you don't have additional block devices, this stat is not set

rootfs, 2022-11-17:
@jichenjc can you lower the verbosity for this line? It is everywhere in the log
```
 container_metric.go:194] cannot extract: block_devices_used
```

jichenjc, 2022-11-17:
um.... let me add one disk to my VM 
but this is container metric, so create a PV and attach to the container then we can see the info?

and this is V4, so should be good? 
klog.V(4).Infof("cannot extract: %s", metric)

rootfs, 2022-11-17:
@sunya-ch do you need `block_devices_used` in the models?

jichenjc, 2022-11-23:
@sunya-ch I found several places have 

we collect `block_devices_used` but not using it ,that's the reason we saw above confusing logs
do we want to either collect the metrics into metric struct or totally remove this `block_devices_used` if it's not used at all?
seems model server didn't use it , do they?

```
        // TO-DO: remove this hard code metric
        labels = append(labels, blockDeviceLabel)    blockDeviceLabel=`block_devices_used` 

```

jichenjc, 2022-12-06:
@sunya-ch do you have any comments on this ? THanks

jichenjc, 2022-12-06:
https://github.com/sustainable-computing-io/kepler/pull/447

as simple process , but not sure it's the right way

sunya-ch, 2023-03-07:
@jichenjc Sorry for delays. Just check the mentioned list. 
I didn't use it in the model for now.

marceloamaral, 2023-05-18:
@jichenjc now that we have a new cgroup lib do you still have this issue?
Can we close this?

jichenjc, 2023-05-18:
I am actually still see this

closed 2023-05-18 by jichenjc