sustainable-computing-io/kepler #362: many PIDs printed in log is confusing

jichenjc, 2022-11-07:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

like this 
```
I1107 01:37:29.093419       1 metric_collector.go:127] energy from pod (1 processes): name: kube-scheduler-jitest40 namespace: kube-system
        cgrouppid: 0 pid: [12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744 12744] 
```

this is pkg/collector/metric/container_metric.go    c.PIDS = append(c.PIDS, pid)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

marceloamaral, 2022-11-07:
This PID list was being used to calculate GPU power consumption and PR #345 removes this dependency.

So, we might remove this

rootfs, 2022-11-07:
@jichenjc nice finding! Can you remove the duplicate values from the PIDS slice [here](https://github.com/sustainable-computing-io/kepler/blob/main/pkg/collector/metric/container_metric.go#L143)?

PR #363 (2022-11-07): fix: avoid put duplicate PID into array
fix #362 
output looks like :(20180 below only occur once)

```
I1107 02:48:35.842949       1 metric_collector.go:127] energy from pod (1 processes): name: prometheus-k8s-0 namespace: monitoring
        cgrouppid: 0 pid: [20180] comm: prometheus
        ePkg (mJ): 0 (0) (eCore: 0 (0) eDram: 0 (0) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0)
        eDyn (mJ): 0 (0)
        avgFreq: 0.00
        CPUTime:  4008 (1358979)
        counters: map[cache_miss:0 (0) cpu_cycles:0 (0) cpu_instr:0 (0)]
        cgroupfs: map[]
        kubelets: map[container_c [... cut]
closed 2022-11-07 by rootfs