sustainable-computing-io/kepler #364: Remove BPF dependency so that Kepler can run without BPF

marceloamaral, 2022-11-07:
**Is your feature request related to a problem? Please describe.**

The bpf collects metrics by processes and maps process ids to container ids.
The structure that contains the container information is `ContainersMetrics`. See [here](https://github.com/sustainable-computing-io/kepler/blob/bf17c6bfa1062e4d557d0aeb1ce2ffc67ade803e/pkg/collector/container_energy_collector.go#L33).

However, currently only the bpf code updates the `ContainersMetrics` map.
So if bpf is not running, the `ContainersMetrics` map will not be updated.

**Describe alternatives you've considered**

We have different approaches to fixing this.
Fix the bpf dependency issue. So that the bpf will always runs. Issue #323 starts this discussion.

Or find other ways to identify the running containers on the system without bpf.

rootfs, 2022-11-15:
Need to identify the use case that doesn't support/need eBPF and then follow up on the container finding case. Check alternative implementations on hardware counters vs e.g. cpu time.

marceloamaral, 2023-05-18:
The https://github.com/sustainable-computing-io/kepler/pull/635 will fix this issue

rootfs, 2023-06-22:
this is fixed now with preinstalled kernel source