sustainable-computing-io/kepler #367: Kepler on VM with Hardware Counters and RAPL

marceloamaral, 2022-11-08:
**Is your feature request related to a problem? Please describe.**
By Default the VM does not expose hardware counters and neither RAPL power metrics.

**Describe the solution you'd like**
In theory, with CPU passthrough, the VM can access hardware counter information related to the VM.

There is a KVM patch that expose RAPL to the VM
[https://lore.kernel.org/all/20220109182317.1075762-1-biche@biche.re/](https://lore.kernel.org/all/20220109182317.1075762-1-biche@biche.re/)
But we need to test it and verify if it is still a patch or it is already officially supported.

**Additional context**
The VM will also need to be configured with bpf libraries. 
Kepler Operator could install the packages when needed...

jichenjc, 2022-11-15:
looks like it's not merged yet
https://github.com/torvalds/linux/blob/master/arch/x86/include/asm/kvm_host.h#L955

rootfs, 2022-11-15:
i doubt if that PR can be merged. For one thing, what value would kvm set for the virtual CPU? There is no per CPU level energy stats...

marceloamaral, 2023-05-18:
Since we don't think that we will be able to collect RAPl metrics inside a VM, I am closing this issue

closed 2023-05-18 by marceloamaral