sustainable-computing-io/kepler #375: difference between kepler-model-server and kepler-estimator

jichenjc, 2022-11-09:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

I am getting all pod energy are 0, check code seems I need at least enable one of kepler-model-server and kepler-estimator
what's the difference between them and which one should use? the side car (https://github.com/sustainable-computing-io/kepler-estimator) ?

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

rootfs, 2022-11-09:
@sunya-ch @KaiyiLiu1234 can you provide a high level usage description? Preferably [here](https://sustainable-computing.io/html/design/index.html)

sunya-ch, 2022-11-09:
I put the general explanation about power estimation and deployment [here](https://sustainable-computing.io/html/design/power_estimate/index.html). 

With the minimum deployment (no estimator and no model server), the kepler should use an offline model weight with embed linear regression model defined by _PodComponentPowerModelConfig_ variable (which is currently hard coded pointing to [this file](https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-server/main/tests/test_models/DynComponentModelWeight/CgroupOnly/ScikitMixed/ScikitMixed.json)
https://github.com/sustainable-computing-io/kepler/blob/1a0f34266c809dbdb6c86419707daf57ff4c1ba6/pkg/model/pod_power.go#L35 

The high level usage description in the document will be coming soon. I will remove the hard-coded variable and do the local test with full integration to the current version of Kepler first.

sunya-ch, 2022-11-09:
@jichenjc As there are possible unmatched features name to the current version of Kepler, the initial model weight may not be applicable. You may further investigate the json file and the collected metric calling below.
https://github.com/sustainable-computing-io/kepler/blob/1a0f34266c809dbdb6c86419707daf57ff4c1ba6/pkg/collector/container_energy_collector.go#L76-L79

rootfs, 2022-11-09:
cc @sallyom @husky-parul @KaiyiLiu1234

jichenjc, 2022-11-10:
>>>I put the general explanation about power estimation and deployment [here](https://sustainable-computing.io/html/design/power_estimate/index.html).

this is great sharing , thanks ! @sunya-ch 

about the original issue, I thinks it's by default there is no model server param provided in manifest 
and this makes following to be come not called 

pkg/collector/metric/utils.go
```
if config.ModelServerEndpoint != "" {
                model.InitEstimateFunctions(names, NodeMetadataNames, NodeMetadataValues)
        }
```

and in turn all the logic reltaed to `PodComponentPowerModelConfig` is not called at all

jichenjc, 2022-11-10:
@marceloamaral we have https://github.com/sustainable-computing-io/kepler/blob/main/pkg/model/model.go#L66
hardcoded ,then 

https://github.com/sustainable-computing-io/kepler/blob/main/cmd/exporter.go#L84 set the endpoint
and the only usage is https://github.com/sustainable-computing-io/kepler/blob/main/pkg/collector/metric/utils.go#L81

so my proposal is to change this to a bool 
https://github.com/sustainable-computing-io/kepler/blob/main/pkg/collector/metric/utils.go#L81

so we have 
- case that doesn't have model server
- case that have model server with different URL
- case that have want to use model server but no URL given, plan default URL (defined above)

so how about make the hard code modelserver URL in manifest or DOC
so anyone want to use model server , input that hard code string or any URL he want 
if no modelserver URL, do no action on model server part, but we won't block sidecar related code even the model server is "" ?

@marceloamaral ?

husky-parul, 2022-11-10:
IMO we should externalize this via a `configmap` if we are not doing already

```yaml
apiVersion: v1
kind: ConfigMap
metadata:
  name: model-server-cfm
  namespace: kepler-system
data:
  ENABLED: (true or false)
  ENDPOINT: ("https://some.endpoint" or "")
  ...
  ```

jichenjc, 2022-11-11:
>>>IMO we should externalize this via a configmap [here](https://github.com/sustainable-computing-io/kepler-model-server/blob/0ac363730011eccf7af07f11a667b525df48412c/manifests/local/deployment.yaml)

but then we depends on model-server ?should this be a resource defined in kepler instead of kepler-model-server?

jichenjc, 2022-11-11:
I can 
- create a configmap like above 
- if no configmap, the default is not read model server (same to not enabled)
- honor the config map when access URL (default will be put into configmap from hard code)

thoughts?

husky-parul, 2022-11-11:
> I can
> 
> * create a configmap like above
> * if no configmap, the default is not read model server (same to not enabled)
Does it mean if no configmap, Kepler will use default model and when configmap is provided by the user it will have models provided by user to be accessed via `URL`
> * honor the config map when access URL (default will be put into configmap from hard code)
> 
> thoughts?

jichenjc, 2022-11-14:
@husky-parul based on my test

>>> Does it mean if no configmap, Kepler will use default model and when configmap is provided by the user it will have models provided by user to be accessed via URL

no endpoint then 
https://github.com/sustainable-computing-io/kepler/blob/main/pkg/collector/metric/utils.go#L81
will block all further operations, 

if it does have some value
https://github.com/sustainable-computing-io/kepler/blob/main/pkg/model/model.go#L30
will be used as hard code ..

so I am curious on the logic here

jichenjc, 2022-11-14:
I proposed 
https://github.com/sustainable-computing-io/kepler/pull/384

rootfs, 2022-11-15:
@sunya-ch will have a discussion on GH project, also going to have this supported in operator

closed 2023-05-24 by stale[bot]