sustainable-computing-io/kepler #394: Provide Community Meeting Invite and Notes

wangchen615, 2022-11-15:
**Is your feature request related to a problem? Please describe.**
We do not yet have a doc to share the community meeting info,  meeting notes, and meeting recordings on this GitHub.

**Describe the solution you'd like**
Provide a community meeting doc to include the above info.

**Describe alternatives you've considered**
Or provide the above info in the wiki page or documentation website.

rootfs, 2022-11-15:
the meeting info is [here](https://github.com/sustainable-computing-io/kepler#community-meetings)

rootfs, 2022-11-15:
@husky-parul would you mind posting the meeting info on kepler-doc?

nikimanoledaki, 2022-12-12:
This could go under the new `Project` section on the docs website 😄

closed 2023-05-24 by stale[bot]