sustainable-computing-io/kepler #417: Inject AbsComponentModelWeight  into kepler LR model?

jichenjc, 2022-11-24:
**Describe the bug**
A clear and concise description of what the bug is.

seeing this  in VM env (not kind)
```
I1124 00:28:09.433823       1 node_energy_collector.go:57] No nodeComponentsEnergy found, node components energy metrics is not exposed
```

this means AbsComponentModelWeight not inited and not able to get estimated node data, looks like no model server initURL 
added (only DynComponentModelWeight have now), 

so the question is it possible that we include those from kepler-model server to kepler LR model?

./tests/test_models/AbsComponentPower/Full/KerasCompFullPipeline/KerasCompFullPipeline.json
./tests/test_models/AbsComponentPower/Full/KerasCompFullPipeline/metadata.json

detail
```
I1124 00:28:06.797549       1 lr.go:167] LR Model (AbsModelWeight): no config
I1124 00:28:06.797556       1 model.go:76] Model AbsModelWeight initiated (false)
I1124 00:28:06.797560       1 lr.go:167] LR Model (AbsComponentModelWeight): no config
I1124 00:28:06.797564       1 model.go:76] Model AbsComponentModelWeight initiated (false)
I1124 00:28:06.797568       1 lr.go:167] LR Model (DynModelWeight): no config
I1124 00:28:06.797571       1 model.go:76] Model DynModelWeight initiated (false)
I1124 00:28:06.797574       1 lr.go:158] getWeightFromServer failed, using InitModelURL:https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-server/main/tests/test_models/DynComponentModelWeight/CgroupOnly/ScikitMixed/ScikitMixed.json to load
I1124 00:28:07.179285       1 lr.go:164] LR Model (DynComponentModelWeight) Weight: map[dram:{{5318.157697416588 map[] map[bytes_read:{0 0 0} bytes_writes:{7.841098405047592e+16 1.0766045667937836e+36 0} cgroupfs_cpu_usage_us:{1.0056068008622782e+07 1.40009411083803e+14 1446.1905640274801} cgroupfs_memory_usage_bytes:{1.944592404177567e+18 1.5694324114250333e+37 140.278995536714} cgroupfs_system_cpu_usage_us:{354378.9946563031 1.3693308434958022e+11 2808.988723422464} cgroupfs_user_cpu_usage_us:{1.0056068008622782e+07 1.40009411083803e+14 0}]}} pkg:{{5326.461712554309 map[] map[bytes_read:{0 0 0} bytes_writes:{7.841098405047592e+16 1.0766045667937836e+36 0} cgroupfs_cpu_usage_us:{1.0056068008622782e+07 1.40009411083803e+14 0} cgroupfs_memory_usage_bytes:{1.944592404177567e+18 1.5694324114250333e+37 149.97387537247224} cgroupfs_system_cpu_usage_us:{354378.9946563031 1.3693308434958022e+11 2814.0186059477487} cgroupfs_user_cpu_usage_us:{1.0056068008622782e+07 1.40009411083803e+14 1452.6093109776657}]}}]
I1124 00:28:07.179352       1 model.go:76] Model DynComponentModelWeight initiated (true)
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

sunya-ch, 2022-12-14:
Please allow me to reopen this issue. 
I think we should not add the default model for the node power for the following reason (extending from the third point to fix in https://github.com/sustainable-computing-io/kepler/issues/461)

- In the VM case, there is no node information (cpu frequency, architecture). Unlike pod/workload power, node power is highly dependent to those information. 
- node power model cannot be relied on only cgroup metric especially with the current kepler collection because it does not collect cgroup stat of the system_processes. Also the current reference model is using full metric which mean the other unavailable metrics such as cpu_cycles from HW counter are zeros and the predicted result can be very inaccurate.

I'm not sure whether should we report the node power in the VM case.
If so, we may need to update kepler to collect the cgroup stat of the VM from the cgroup top path and use that value to estimate the node power with overridden cpu frequency and architecture from node profile.

rootfs, 2022-12-14:
sounds good @sunya-ch. For VMs that don't have host cpu passthrough, predicting node energy is not reliable.  

Just a thought, does it make sense to use another categorical parameter using cpu governor (powersave/performance) and cpu utilization (since they determine cpu frequency setting by kernel) in the node model (assuming temperature is not a factor)?

marceloamaral, 2022-12-14:
ebpf should report the cpu time of `system_processes`, so it should be possible to predict the `system_processes` within the VM.
The node power of the VM, should be the sum of all containers + system_processes.

sunya-ch, 2023-03-07:
> @marceloamaral that sounds a good plan. If I guess in this case, we use the regression based model to get the per container energy, and aggregate them to get the node energy. @sunya-ch does the regression model depend on the node energy?

Currently-used model is based on only RAPL.

marceloamaral, 2023-05-18:
@sunya-ch can you follow up this PR?

sunya-ch, 2023-06-21:
I'm sorry but I suggest to pending this PR until we have node-level models ready (with node profiles).

PR #420 (2022-11-27): fix: add default node component power estimator
fix #417
closed 2023-08-28 by stale[bot]