sustainable-computing-io/kepler #418: Remove duplicated Kepler Installation steps

nikimanoledaki, 2022-11-24:
Hi, Kepler community! 👋  Small docs UX enhancement proposal!

The Kepler Installation steps are in this repo's README and they are also in the `kepler-docs` repo / website, [here](https://github.com/sustainable-computing-io/kepler-doc/blob/main/docs/installation/kepler.md).

The sections duplicated are from [Requirement](https://github.com/sustainable-computing-io/kepler#requirement) to [Configure Prometheus to scrape Kepler-exporter endpoints](https://github.com/sustainable-computing-io/kepler#configure-prometheus-to-scrape-kepler-exporter-endpoints).

I would like to remove the duplicated sections from this repo and replace them with a link that points to the Kepler website in order to make that the single source of truth.

Does this sound ok to you? If it does, I will take this action. Thanks :)

jichenjc, 2022-11-25:
I saw various projects do this as xxx-docs are for public new comer to whole project and 
github readme is for project itself startup

maybe we can create a link if you think they might be inconsistent?

nikimanoledaki, 2022-11-25:
Ah, I see what you mean :) If I understand correctly, there should be 2 separate installation methods for Kepler:
- Developer installation (in the `kepler` repo)
- End-user installation (in the `kepler-docs` repo / website)

And Helm coming soon :)

Here are some suggestions:
- Move the developer installation from the `kepler` repo's README to [Developer.md](https://github.com/sustainable-computing-io/kepler/blob/main/Developer.md). Add a link in the README. Rename `Developer.md` to `CONTRIBUTING.md` to follow the CNCF recommendations ([here](https://contribute.cncf.io/maintainers/github/templates/required/contributing/)).
- Simplify the end-user installation steps in the `kepler-doc` repo so that users are not required to clone the repo or do a `make` command in order to install Kepler's K8s manifests.
- Add a link to the `kepler` repo's README that links to the simpler installations steps in `kepler-docs`.

What do you think?

jichenjc, 2022-11-26:
I am ok with the proposal,  @rootfs @marceloamaral any comments?

SamYuan1990, 2022-11-26:
LGTM.

rootfs, 2022-11-27:
@nikimanoledaki sounds good!

SamYuan1990, 2023-05-16:
@nikimanoledaki  or @rootfs are we able to close this one as the 3rd task is out of date?

closed 2023-05-16 by rootfs