sustainable-computing-io/kepler #422: 0 processes reported in log

jichenjc, 2022-11-25:
**Describe the bug**
A clear and concise description of what the bug is.

VM env:

```
I1125 01:48:29.381448       1 metric_collector.go:132] energy from pod (0 processes): name: blackbox-exporter-59cccb5797-wsqsq namespace: monitoring
I1125 01:48:29.381488       1 metric_collector.go:132] energy from pod (0 processes): name: prometheus-k8s-1 namespace: monitoring
I1125 01:48:29.381623       1 metric_collector.go:132] energy from pod (0 processes): name: kube-state-metrics-6d68f89c45-xbjsv namespace: monitoring
I1125 01:48:29.381723       1 metric_collector.go:132] energy from pod (0 processes): name: alertmanager-main-1 namespace: monitoring
I1125 01:48:29.381757       1 metric_collector.go:132] energy from pod (0 processes): name: kube-state-metrics-6d68f89c45-xbjsv namespace: monitoring
I1125 01:48:32.390264       1 metric_collector.go:132] energy from pod (0 processes): name: blackbox-exporter-59cccb5797-wsqsq namespace: monitoring
I1125 01:48:32.390327       1 metric_collector.go:132] energy from pod (0 processes): name: prometheus-k8s-1 namespace: monitoring
I1125 01:48:32.390348       1 metric_collector.go:132] energy from pod (0 processes): name: alertmanager-main-0 namespace: monitoring
I1125 01:48:32.390367       1 metric_collector.go:132] energy from pod (0 processes): name: prometheus-k8s-0 namespace: monitoring
I1125 01:48:32.390517       1 metric_collector.go:132] energy from pod (0 processes): name: kube-state-metrics-6d68f89c45-xbjsv namespace: monitoring
I1125 01:48:32.390608       1 metric_collector.go:132] energy from pod (0 processes): name: node-exporter-fsgcl namespace: monitoring
I1125 01:48:32.390672       1 metric_collector.go:132] energy from pod (0 processes): name: alertmanager-main-1 namespace: monitoring
I1125 01:48:32.390719       1 metric_collector.go:132] energy from pod (0 processes): name: prometheus-k8s-1 namespace: monitoring
I1125 01:48:32.390747       1 metric_collector.go:132] energy from pod (0 processes): name: kube-state-metrics-6d68f89c45-xbjsv namespace: monitoring
I1125 01:48:32.390814       1 metric_collector.go:132] energy from pod (0 processes): name: node-exporter-fsgcl namespace: monitoring
I1125 01:48:32.390855       1 metric_collector.go:132] energy from pod (0 processes): name: prometheus-operator-67f59d65b8-x6q7v namespace: monitoring
I1125 01:48:32.391008       1 metric_collector.go:132] energy from pod (0 processes): name: kube-proxy-zs9t6 namespace: kube-system
I1125 01:48:32.391033       1 metric_collector.go:132] energy from pod (0 processes): name: alertmanager-main-0 namespace: monitoring
I1125 01:48:32.391075       1 metric_collector.go:132] energy from pod (0 processes): name: blackbox-exporter-59cccb5797-wsqsq namespace: monitoring
I1125 01:48:32.391116       1 metric_collector.go:132] energy from pod (0 processes): name: blackbox-exporter-59cccb5797-wsqsq namespace: monitoring
I1125 01:48:32.391138       1 metric_collector.go:132] energy from pod (0 processes): name: prometheus-k8s-0 namespace: monitoring

```
though some pods are in crash, but most of them are active running , why we reported 0 processes ??
```
[root@jitest42 ~]# kubectl get pods -A
NAMESPACE     NAME                                         READY   STATUS             RESTARTS           AGE
kepler        kepler-exporter-5cpcw                        1/1     Running            0                  11m
kube-system   coredns-565d847f94-9s84n                     1/1     Running            2                  17d
kube-system   coredns-565d847f94-bmlmf                     1/1     Running            2                  17d
kube-system   etcd-jitest42.novalocal                      1/1     Running            2                  17d
kube-system   kube-apiserver-jitest42.novalocal            1/1     Running            2                  17d
kube-system   kube-controller-manager-jitest42.novalocal   1/1     Running            2                  17d
kube-system   kube-proxy-zs9t6                             1/1     Running            2                  17d
kube-system   kube-scheduler-jitest42.novalocal            1/1     Running            2                  17d
monitoring    alertmanager-main-0                          1/2     CrashLoopBackOff   3872 (2m33s ago)   11d
monitoring    alertmanager-main-1                          1/2     Running            3872 (6m23s ago)   11d
monitoring    alertmanager-main-2                          1/2     CrashLoopBackOff   3873 (3m53s ago)   11d
monitoring    blackbox-exporter-59cccb5797-wsqsq           3/3     Running            0                  11d
monitoring    grafana-76d67c8b47-57bn4                     1/1     Running            0                  11d
monitoring    kube-state-metrics-6d68f89c45-xbjsv          3/3     Running            0                  11d
monitoring    node-exporter-fsgcl                          2/2     Running            0                  11d
monitoring    prometheus-adapter-757f9b4cf9-7xqxg          1/1     Running            0                  11d
monitoring    prometheus-adapter-757f9b4cf9-fr576          1/1     Running            0                  11d
monitoring    prometheus-k8s-0                             2/2     Running            0                  11d
monitoring    prometheus-k8s-1                             2/2     Running            0                  11d
monitoring    prometheus-operator-67f59d65b8-x6q7v         2/2     Running            0                  11d

```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

jichenjc, 2022-11-29:
take this as example:

```
I1129 09:39:10.841974       1 metric_collector.go:132] energy from pod/container (0 processes): name: kube-state-metrics-6d68f89c45-xbjsv/kube-rbac-proxy-main namespace: monitoring
        cgrouppid: 0 pid: [] comm:
        ePkg (mJ): 1323 (1323) (eCore: 1323 (1323) eDram: 1330 (1330) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0)
        eDyn (mJ): 0 (0)
        avgFreq: 0.00
        CPUTime:  0 (0)
        counters: map[cache_miss:0 (0) cpu_cycles:0 (0) cpu_instr:0 (0)]
        cgroupfs: map[cgroupfs_cpu_usage_us:0 (401324190) cgroupfs_ioread_bytes:0 (0) cgroupfs_iowrite_bytes:0 (0) cgroupfs_memory_usage_bytes:0 (25673728) cgroupfs_system_cpu_usage_us:0 (176419080) cgroupfs_user_cpu_usage_us:0 (224905109)]
        kubelets: map[container_cpu_usage_seconds_total:0 (401) container_memory_working_set_bytes:0 (25673728)]
```

it's reset from 
```
I1129 09:39:10.544729       1 container_metric.go:123] ---reset--kube-state-metrics-6d68f89c45-xbjsv--kube-rbac-proxy-main

```

but not reach container_hc_collector.go:121 (which do the CurrProcess++) , this means 
no process or container read from eBPF at al...

marceloamaral, 2022-11-29:
Is BPF running?

jichenjc, 2022-11-30:
I think so as I have one log 
```
I1130 00:11:44.296708       1 bcc_attacher.go:131] Successfully load eBPF module with option: [-DNUM_CPUS=8 -DCPU_FREQ -DSET_GROUP_ID]
```

rootfs, 2022-11-30:
@jichenjc can you post the beginning of all the logs till `exporter.go:202] Started Kepler `?

jichenjc, 2022-11-30:
```
I1130 01:00:08.032892       1 exporter.go:149] Kepler running on version: v0.3-210-g280152f-dirty
I1130 01:00:08.032983       1 config.go:124] using gCgroup ID in the BPF program: true
I1130 01:00:08.033004       1 config.go:125] kernel version: 4.18
I1130 01:00:08.033040       1 slice_handler.go:145] InitSliceHandler: &{map[] /sys/fs/cgroup/kubepods.slice /sys/fs/cgroup/kubepods.slice /sys/fs/cgroup/kubepods.slice}
I1130 01:00:08.033059       1 bcc_attacher.go:143] hardeware counter metr true
I1130 01:00:08.059116       1 bcc_attacher.go:143] hardeware counter metr true
I1130 01:00:08.059161       1 utils.go:38] Available counter metrics: [cpu_cycles cpu_instr cache_miss]
I1130 01:00:08.059190       1 utils.go:39] Available cgroup metrics from cgroup: [cgroupfs_cpu_usage_us cgroupfs_system_cpu_usage_us cgroupfs_user_cpu_usage_us cgroupfs_ioread_bytes cgroupfs_iowrite_bytes cgroupfs_memory_usage_bytes]
I1130 01:00:08.059206       1 utils.go:40] Available cgroup metrics from kubelet: [container_cpu_usage_seconds_total container_memory_working_set_bytes]
I1130 01:00:08.059210       1 utils.go:41] Available I/O metrics: [bytes_read bytes_writes]
I1130 01:00:08.059218       1 lr.go:167] LR Model (AbsModelWeight): no config
I1130 01:00:08.059224       1 model.go:76] Model AbsModelWeight initiated (false)
I1130 01:00:08.059227       1 lr.go:158] getWeightFromServer failed, using InitModelURL:https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-server/main/tests/test_models/AbsComponentModelWeight/Full/KerasCompWeightFullPipeline/KerasCompWeightFullPipeline.json to load
I1130 01:00:08.417902       1 lr.go:164] LR Model (AbsComponentModelWeight) Weight: map[core:{{17.49994468688965 map[cpu_architecture:map[Alder Lake:{0.5408945679664612} Broadwell:{17.9639892578125} Cascade Lake:{-0.49166440963745117} Coffee Lake:{0.5166589617729187} Haswell:{-0.5789095163345337} Ivy Bridge:{-0.024028241634368896} Sandy Bridge:{0.5239214301109314} Sky Lake:{0.4193417429924011}]] map[cpu_cycles:{6.85713664e+09 7.560771917192364e+18 -0.11352460086345673} cpu_instr:{3.374244864e+09 8.408530291701842e+17 -0.414739191532135} cpu_time:{192019.5 5.2761312e+08 -0.06457684189081192}]}} dram:{{17.49994468688965 map[cpu_architecture:map[Alder Lake:{-0.11559933423995972} Broadwell:{16.972564697265625} Cascade Lake:{0.5505847334861755} Coffee Lake:{-0.4564790725708008} Haswell:{-0.13912856578826904} Ivy Bridge:{-0.018331050872802734} Sandy Bridge:{-0.6695247888565063} Sky Lake:{0.29698115587234497}]] map[cache_miss:{9.329199e+06 2.2145245642752e+13 -0.4680119752883911} container_memory_working_set_bytes:{253952 1.93474854912e+11 0.6805805563926697}]}}]
I1130 01:00:08.418001       1 model.go:76] Model AbsComponentModelWeight initiated (true)
I1130 01:00:08.418011       1 lr.go:167] LR Model (DynModelWeight): no config
I1130 01:00:08.418015       1 model.go:76] Model DynModelWeight initiated (false)
I1130 01:00:08.418019       1 lr.go:158] getWeightFromServer failed, using InitModelURL:https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-server/main/tests/test_models/DynComponentModelWeight/CgroupOnly/ScikitMixed/ScikitMixed.json to load
I1130 01:00:08.418015       1 model.go:76] Model DynModelWeight initiated (false)
I1130 01:00:08.418019       1 lr.go:158] getWeightFromServer failed, using InitModelURL:https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-server/main/tests/test_models/DynComponentModelWeight/CgroupOnly/ScikitMixed/ScikitMixed.json to load
I1130 01:00:08.575966       1 lr.go:164] LR Model (DynComponentModelWeight) Weight: map[dram:{{5318.157697416588 map[] map[bytes_read:{0 0 0} bytes_writes:{7.841098405047592e+16 1.0766045667937836e+36 0} cgroupfs_cpu_usage_us:{1.0056068008622782e+07 1.40009411083803e+14 1446.1905640274801} cgroupfs_memory_usage_bytes:{1.944592404177567e+18 1.5694324114250333e+37 140.278995536714} cgroupfs_system_cpu_usage_us:{354378.9946563031 1.3693308434958022e+11 2808.988723422464} cgroupfs_user_cpu_usage_us:{1.0056068008622782e+07 1.40009411083803e+14 0}]}} pkg:{{5326.461712554309 map[] map[bytes_read:{0 0 0} bytes_writes:{7.841098405047592e+16 1.0766045667937836e+36 0} cgroupfs_cpu_usage_us:{1.0056068008622782e+07 1.40009411083803e+14 0} cgroupfs_memory_usage_bytes:{1.944592404177567e+18 1.5694324114250333e+37 149.97387537247224} cgroupfs_system_cpu_usage_us:{354378.9946563031 1.3693308434958022e+11 2814.0186059477487} cgroupfs_user_cpu_usage_us:{1.0056068008622782e+07 1.40009411083803e+14 1452.6093109776657}]}}]
I1130 01:00:08.576026       1 model.go:76] Model DynComponentModelWeight initiated (true)
I1130 01:00:08.576035       1 exporter.go:164] Initializing the GPU collector
perf_event_open: No such file or directory
I1130 01:00:10.978569       1 bcc_attacher.go:90] failed to attach perf event cpu_cycles: failed to open bpf perf event: no such file or directory
perf_event_open: No such file or directory
I1130 01:00:10.978639       1 bcc_attacher.go:90] failed to attach perf event cpu_instr: failed to open bpf perf event: no such file or directory
perf_event_open: No such file or directory
I1130 01:00:10.978699       1 bcc_attacher.go:90] failed to attach perf event cache_miss: failed to open bpf perf event: no such file or directory
I1130 01:00:10.978715       1 bcc_attacher.go:131] Successfully load eBPF module with option: [-DNUM_CPUS=8 -DCPU_FREQ -DSET_GROUP_ID]
I1130 01:00:10.996339       1 node_energy_collector.go:54] Node components power model collection is supported
I1130 01:00:10.997136       1 exporter.go:202] Started Kepler in 2.96429336s
```

jichenjc, 2022-11-30:
ok, I will check what I can do at the place you pointed out,actually ,seems the cgroup ID is not calculated correctly (I added some logs in GetContainerID function , will paste update when I have more clear answer

jichenjc, 2022-12-01:
take following as example, the cgroup ID not reported every loop of updateBPFMetrics
sometime it reports then we are able to get container info from the cgroup BUT sometimes it does not

see below, I print from HC collector logs and see 
I1201 00:41:00.405510  ==> this time it reports 
I1201 00:41:03.141720  ==> this time no report in eBPF layer

is it because it's inactive sometime so eBPF not able to find it ?

```
I1201 00:41:00.181040       1 container_hc_collector.go:122] ----kube-scheduler-jitest42.novalocal---kube-scheduler--0
I1201 00:41:00.181098       1 resolve_container.go:193] getContainerIDFromPath with 4295008359:277829:true
I1201 00:41:00.181142       1 container_hc_collector.go:81] ==7b1d84ec7c2a857c92eee168ea9884a082969d755fb5500886f25a784e03efec==4295008359
I1201 00:41:00.181180       1 container_hc_collector.go:122] ----node-exporter-fsgcl---kube-rbac-proxy--0
I1201 00:41:00.181238       1 resolve_container.go:193] getContainerIDFromPath with 4295011629:278755:true
I1201 00:41:00.181272       1 container_hc_collector.go:81] ==system_processes==4295011629
--
        kubelets: map[container_cpu_usage_seconds_total:0 (574) container_memory_working_set_bytes:0 (23068672)]

I1201 00:41:00.405510       1 metric_collector.go:132] energy from pod/container (1 processes): name: node-exporter-fsgcl/kube-rbac-proxy namespace: monitoring
        cgrouppid: 4295008359 pid: [277829] comm: kube-rbac-proxy
        ePkg (mJ): 1323 (1323) (eCore: 1323 (1323) eDram: 1330 (1330) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0)
        eDyn (mJ): 0 (0)
        avgFreq: 0.00
--
        kubelets: map[container_cpu_usage_seconds_total:0 (55003) container_memory_working_set_bytes:0 (439934976)]

I1201 00:41:03.141720       1 metric_collector.go:132] energy from pod/container (0 processes): name: node-exporter-fsgcl/kube-rbac-proxy namespace: monitoring
        cgrouppid: 4295008359 pid: [277829] comm: kube-rbac-proxy
        ePkg (mJ): 1323 (2646) (eCore: 1323 (2646) eDram: 1330 (2660) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0)
        eDyn (mJ): 0 (0)
        avgFreq: 0.00
--
        kubelets: map[container_cpu_usage_seconds_total:0 (55003) container_memory_working_set_bytes:0 (439934976)]

I1201 00:41:06.149213       1 metric_collector.go:132] energy from pod/container (0 processes): name: node-exporter-fsgcl/kube-rbac-proxy namespace: monitoring
        cgrouppid: 4295008359 pid: [277829] comm: kube-rbac-proxy
        ePkg (mJ): 1323 (3969) (eCore: 1323 (3969) eDram: 1330 (3990) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0)
        eDyn (mJ): 0 (0)
        avgFreq: 0.00
--
```

jichenjc, 2022-12-01:
```
--
I1201 00:41:09.170386       1 resolve_container.go:193] getContainerIDFromPath with 4294975170:2886262:true
I1201 00:41:09.170391       1 container_hc_collector.go:81] ==system_processes==4294975170
I1201 00:41:09.170396       1 container_hc_collector.go:122] ----system_processes---system_processes--15
I1201 00:41:09.170405       1 resolve_container.go:193] getContainerIDFromPath with 4295008359:277829:true
I1201 00:41:09.170409       1 container_hc_collector.go:81] ==7b1d84ec7c2a857c92eee168ea9884a082969d755fb5500886f25a784e03efec==4295008359
I1201 00:41:09.170415       1 container_hc_collector.go:122] ----node-exporter-fsgcl---kube-rbac-proxy--0
I1201 00:41:09.170490       1 resolve_container.go:193] getContainerIDFromPath with 4294967297:577:true
I1201 00:41:09.170507       1 container_hc_collector.go:81] ==system_processes==4294967297
--
        kubelets: map[container_cpu_usage_seconds_total:0 (55003) container_memory_working_set_bytes:233472 (440168448)]

I1201 00:41:09.219138       1 metric_collector.go:132] energy from pod/container (1 processes): name: node-exporter-fsgcl/kube-rbac-proxy namespace: monitoring
        cgrouppid: 4295008359 pid: [277829] comm: kube-rbac-proxy
        ePkg (mJ): 1323 (5292) (eCore: 1323 (5292) eDram: 1330 (5320) eUncore: 0 (0)) eGPU (mJ): 0 (0) eOther (mJ): 0 (0)
        eDyn (mJ): 0 (0)
```

it can be found again after a few seconds ...

rootfs, 2022-12-01:
I have some inactive processes on my environment as well, they are always flannel and kube-proxy:
```console
# kubectl logs -n kepler daemonset/kepler-exporter -f |grep "(0 processes" |head 
I1201 00:59:10.732555       1 metric_collector.go:132] energy from pod (0 processes): name: kube-flannel-ds-ldkvn namespace: kube-flannel 
I1201 00:59:10.732721       1 metric_collector.go:132] energy from pod (0 processes): name: kube-flannel-ds-ldkvn namespace: kube-flannel 
I1201 00:59:13.660756       1 metric_collector.go:132] energy from pod (0 processes): name: kube-flannel-ds-ldkvn namespace: kube-flannel 
I1201 00:59:13.660791       1 metric_collector.go:132] energy from pod (0 processes): name: kube-flannel-ds-ldkvn namespace: kube-flannel 
I1201 00:59:13.660901       1 metric_collector.go:132] energy from pod (0 processes): name: kube-flannel-ds-ldkvn namespace: kube-flannel 
I1201 00:59:16.658832       1 metric_collector.go:132] energy from pod (0 processes): name: kube-flannel-ds-ldkvn namespace: kube-flannel 
I1201 00:59:16.658844       1 metric_collector.go:132] energy from pod (0 processes): name: kube-flannel-ds-ldkvn namespace: kube-flannel 
I1201 00:59:19.655789       1 metric_collector.go:132] energy from pod (0 processes): name: kube-proxy-967tr namespace: kube-system 
I1201 00:59:19.655865       1 metric_collector.go:132] energy from pod (0 processes): name: kube-flannel-ds-ldkvn namespace: kube-flannel 
I1201 00:59:19.656080       1 metric_collector.go:132] energy from pod (0 processes): name: kube-flannel-ds-ldkvn namespace: kube-flannel 
```

jichenjc, 2022-12-01:
so this is desired behavior? if so we can close this as reference (or maybe add some doc explain this to avoid confusion for further questions)

rootfs, 2022-12-01:
you can make the logging more meaningful, e.g. `active processes` instead of `processes`

PR #442 (2022-12-01): fix: update log to avoid confusion (0 process) ==> (0 active process)
Fixes: #422
closed 2022-12-01 by rootfs