sustainable-computing-io/kepler #424: how to do ratio calculation correctly in local lr?

jichenjc, 2022-11-28:
**Describe the bug**
A clear and concise description of what the bug is.

1) question on containerNumber
containerNumber => this include system process, should we exclude it?

https://github.com/sustainable-computing-io/kepler/blob/main/pkg/model/estimator/local/ratio.go#L52

2) question on algorithm

as comment said `power = nodeResEnergyUtilization / containerNumber` is not a good way

should we consider cgroup data e.g the (cgroupfs_cpu_usage_us) / total?
```
cgroupfs_user_cpu_usage_us cgroupfs_ioread_bytes cgroupfs_iowrite_bytes cgroupfs_memory_usage_bytes cgroupfs_cpu_usage_us cgroupfs_system_cpu_usage_us
```

marceloamaral, 2022-11-28:
Yes, `cgroupfs_cpu_usage_us` is a good metric candidate

jichenjc, 2022-11-28:
ok, will see how we can improve here

closed 2023-05-24 by stale[bot]