sustainable-computing-io/kepler #444: some node metrics missing label

SamYuan1990, 2022-12-02:
kepler_node_nodeInfo
kepler_node_energy_stat
kepler_node_package_energy_millijoule

```
# HELP kepler_node_nodeInfo Labeled node information
# TYPE kepler_node_nodeInfo counter
kepler_node_nodeInfo{cpu_architecture="unknown"} 1

# HELP kepler_node_energy_stat Several labeled node metrics
# TYPE kepler_node_energy_stat counter
kepler_node_energy_stat{cpu_architecture="unknown",node_block_devices_used="0",node_curr_bytes_read="0",node_curr_bytes_writes="0",node_curr_cache_miss="0",node_curr_container_cpu_usage_seconds_total="0",node_curr_container_memory_working_set_bytes="0",node_curr_cpu_cycles="0",node_curr_cpu_instr="0",node_curr_cpu_time="0",node_curr_energy_in_core_joule="0",node_curr_energy_in_dram_joule="0",node_curr_energy_in_gpu_joule="0",node_curr_energy_in_other_joule="0",node_curr_energy_in_pkg_joule="0",node_curr_energy_in_uncore_joule="0",node_name="kind-control-plane"} 0

# HELP kepler_node_package_energy_millijoule Aggregated RAPL value in package (socket) in milijoules (deprecated)
# TYPE kepler_node_package_energy_millijoule counter
kepler_node_package_energy_millijoule{core="0",dram="1",instance="kind-control-plane",pkg_id="0",uncore="0"} 0
```

ref https://github.com/sustainable-computing-io/kepler/actions/runs/3584038952/jobs/6030200173#step:10:475

jichenjc, 2022-12-07:
compare to the other metric in the same link you provided, I don't know where is the `missing label` you are talking about? @SamYuan1990 ?
```
# HELP kepler_node_package_joules_total Aggregated RAPL value in package (socket) in joules
# TYPE kepler_node_package_joules_total counter
kepler_node_package_joules_total{instance="kind-control-plane",package="0",source="rapl"} 0.019
```

SamYuan1990, 2022-12-07:
@jichenjc it seems for the metrics I listed, missing instance label to identify the instance.
`instance="kind-control-plane",`

marceloamaral, 2022-12-07:
@SamYuan1990 where did you get the metric?
From Prometheus or from the Kepler instance?

SamYuan1990, 2022-12-07:
@marceloamaral , from the Kepler instance.

rootfs, 2022-12-07:
the metrics reported by prometheus provides `instance`.

SamYuan1990, 2022-12-07:
> https://github.com/sustainable-computing-io/kepler/actions/runs/3584038952/jobs/6030200173#step:10:475

do we have any update in recent? as it conflicts with CI output?
I am now rerun a testing for verify.

SamYuan1990, 2022-12-07:
ok, verified via https://github.com/sustainable-computing-io/kepler/pull/453 it seems fixed.

closed 2022-12-07 by SamYuan1990