sustainable-computing-io/kepler #456: e2e tests for Kepler, estimator, and model server

rootfs, 2022-12-09:
**Is your feature request related to a problem? Please describe.**
Having all of the components e2e tested on baremetal and VM (especially on CI)

**Describe the solution you'd like**
The tests should verify that:
- [ ] all the components are configured correctly, up and running
- [ ] the models (ebpf, cgroup, etc) can be trained and updated online

jichenjc, 2022-12-12:
some proposal 

- seems CI result all 0 now (at least some of them)
https://github.com/sustainable-computing-io/kepler/actions/runs/3660745079/jobs/6188232690
so we need fix this first

- maybe overtime we can check 1 min consumption and for all `sum` metrics should see a `>=` change ?

- asked in https://github.com/sustainable-computing-io/kepler/issues/375, seems model server and side car are mutual exclusive?
if so, maybe in CI we can run integ-model-server and integ-side-car test ?

sunya-ch, 2022-12-12:
> some proposal
> 
> * asked in [difference between kepler-model-server and kepler-estimator #375](https://github.com/sustainable-computing-io/kepler/issues/375), seems model server and side car are mutual exclusive?
>   if so, maybe in CI we can run integ-model-server and integ-side-car test ?

Yes, it is mutual exclusive. 
We can have the deployment scenario as described here: https://sustainable-computing.io/design/power_estimation/#deployment-scenarios. 
- minimum (no sidecar, no model server) -> kepler local estimator will use LR model weight downloaded by initial URL.
- with sidecar only -> kepler exporter requests estimated power from the sidecar. The sidecar estimator will use achieved (currently GBR) model downloaded by initial URL.
- with model server only -> kepler local estimator will request LR model weight from model server.
- with both (full deployment) -> kepler exporter requests estimated power from the sidecar. The sidecar estimator will request achieved model from model server.

rootfs, 2022-12-12:
@jichenjc Pods metrics are ok, but the node metrics are currently all zero, we'll deploy a node power model in the CI

sunya-ch, 2022-12-13:
Idea to create e2e tests for integration.

- make build-manifest
- make build-manifest OPTS="ESTIMATOR_SIDECAR_DEPLOY"
    To check whether sidecar is properly set, according to https://github.com/sustainable-computing-io/kepler/issues/461, check log line:
    ```bash
    Model DynComponentPower initiated (true)
    ```
- make build-manifest OPTS="MODEL_SERVER_DEPLOY"
    To check whether model-sever is connected, according to https://github.com/sustainable-computing-io/kepler/issues/463, check log line:
    ```bash
    LR Model (AbsComponentModelWeight): getWeightFromServer: map[...
    ```
- make build-manifest OPTS="ESTIMATOR_SIDECAR_DEPLOY MODEL_SERVER_DEPLOY"
    To check whether both connected, 
    1. check sidecar is properly set
    2. check model-server is connected sidecar (TBD)

jichenjc, 2022-12-14:
>>>check log line: 

even though we can check the logs but seems my understanding is integration test is mostly a focus on functions 
instead of logs ,so the system should work like a black box.. so instead of checking those logs, is there anyway
we can expose some endpoint (like debug endpoint) to curl and see whether the functions works well?

sunya-ch, 2022-12-14:
> even though we can check the logs but seems my understanding is integration test is mostly a focus on functions instead of logs ,so the system should work like a black box.. so instead of checking [...]

I think it is a good ise. we can expose all the status of kepler not only the success/failure of the connection and model config but also the available metrics from its discovery. Then we can let operator use these endpoint to update the Kepler CR status.

closed 2023-05-24 by stale[bot]