sustainable-computing-io/kepler #474: containerIDToContainerInfo should be updated to reflect removed container?

jichenjc, 2022-12-16:
**Describe the bug**
A clear and concise description of what the bug is.

https://github.com/sustainable-computing-io/kepler/blob/main/pkg/cgroup/resolve_container.go#L56

seems this defined somewhere and updated when creation
seems no place to update it when pod got destroyed?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

marceloamaral, 2023-05-02:
The PR #635 might improve this.

sunya-ch, 2023-06-22:
Agree to @marceloamaral.
@jichenjc  Would you mind closing this issue?

rootfs, 2023-06-22:
should be fixed in #635

closed 2023-06-22 by rootfs