sustainable-computing-io/kepler #478: Consume hardware power metrics from Hardware Sentry

bertysentry, 2022-12-20:
**Is your feature request related to a problem? Please describe.**
Solutions exist to collect power metrics, and even semantic conventions. It would be nice if Kepler would leverage that.

**Describe the solution you'd like**
Example of a solution that collects hardware power metrics: [Hardware Sentry](https://www.sentrysoftware.com/products/hardware-sentry.html). It's free but it's **not yet** open-source.

It would be greatly beneficial to Kepler if it could use Hardware Sentry as a source for hardware power metrics (notably `hw_host_energy_joules_total` and `hw_energy_joules_total{hw_type="cpu|gpu|memory|physical_disk|network"}`).

Also, OpenTelemetry have defined [semantic conventions for hardware](https://opentelemetry.io/docs/reference/specification/metrics/semantic_conventions/hardware-metrics/), including for power and energy metrics. Kepler should follow these conventions.

**Describe alternatives you've considered**
None, really.

**Additional context**
I am the CEO of the company who develops Hardware Sentry. We're pushing for solutions that help companies reduce the carbon footprint of their data centers (notably with temperature optimization). I'm very happy to discover Kepler and sustainable-computing-io!

jichenjc, 2022-12-20:
@rootfs @marceloamaral FYI

rootfs, 2022-12-20:
@bertysentry thank for sharing the info. I've checked Hardware Sentry before, it has a nice collection of power sensors. Anything plan in your mind to collaborate?

bertysentry, 2022-12-20:
@rootfs We can definitely work together! We've been wondering for some time how we could derive our physical metrics into the software world, notably VMs and containers. Looks like you guys are doing exactly that, it's awesome!

We could help you setup Hardware Sentry in your environment so you guys can see for yourselves the metrics we're collecting, the attributes, the limitations, etc. (btw version 3.0.00 of Hardware Sentry will be published in a few days and it simplifies the setup process greatly). 

Then you guys could help us setup your software in our environment. We could even integrate it in our [demo platform](https://hws-demo.sentrysoftware.com/d/-GV2ChOnz/hardware-sentry-main?from=now-24h&to=now&orgId=1) (a Grafana instance running public, showing Prometheus metrics collected with Hardware Sentry).

Mixing our metrics with Kepler's metrics would greatly help assess the carbon footprint of an application (leveraging the `hw_site_pue_ratio` and `hw_site_carbon_density_grams` metrics).

We also have some experience with OpenTelemetry (we've been contributing there), so we could help there too!

rootfs, 2022-12-21:
@bertysentry sounds, would you be interested in joining our community meeting and discussing it? The 2023 meeting calendar will posted [here](https://github.com/sustainable-computing-io/kepler#community-meetings) in early Jan.

bertysentry, 2022-12-22:
Awesome, thank you @rootfs!

rootfs, 2023-01-03:
@bertysentry please find our meeting schedule [here](https://github.com/sustainable-computing-io/kepler/pull/484)

closed 2023-05-24 by stale[bot]