sustainable-computing-io/kepler #480: why use dummy Impl of power component instead of estimated?

jichenjc, 2022-12-21:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

we are using dummy if no RAPL or MSR 
https://github.com/sustainable-computing-io/kepler/blob/main/pkg/power/components/power.go#L60

but we do have estimated
https://github.com/sustainable-computing-io/kepler/blob/main/pkg/power/components/source/estimate.go

so from name, same estimated is suitable than dummy? 
**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

jichenjc, 2022-12-22:
@marceloamaral do you know why we have use dummy? Thanks

marceloamaral, 2022-12-22:
I think the dummy was created before for testing purposes.
If I remember correctly, the estimate needs to have the estimate sidecar, right?
Therefore, if we need to run a test without the estimated sidecar and the node does not support RAPL, the dummy will be used.

jichenjc, 2022-12-22:
dummy is selected unconditionally 
let me see whether I can add a estimated logic here (first estimate then if not possible, use dummy)

marceloamaral, 2023-05-18:
@jichenjc is this still a problem?

jichenjc, 2023-05-18:
no, should be ok now

closed 2023-05-18 by jichenjc