sustainable-computing-io/kepler #482: ARM support

rootfs, 2023-01-03:
Some ARM platforms provide power readings, such as [Ampere](https://github.com/AmpereComputing/ampere-lts-kernel/wiki/hwmon-drivers#altra_hwmon), and [Tegra](https://docs.nvidia.com/jetson/archives/l4t-archived/l4t-325/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/power_management_jetson_xavier.html#wwpID0E0AG0HA)

rootfs, 2023-01-24:
Discussion:
1) Should Apple Silicon be supported? Does it expose power reading (tools available) ? Can kepler be started natively on Mac?
2) Use case: ML

odidev, 2023-05-24:
Hi Team,

I have tried deploying Kepler over the arm64 platform on Minikube using Helm. But the Kepler pod fails because the Kepler image is not available for the arm64 platform. 

I successfully built the [docker images](https://hub.docker.com/repository/docker/odidev/kepler/tags?page=1&ordering=last_updated) for arm64 platforms.  With this docker image pods are successfully running.
```
subham@ip-xyz:~$ kubectl get pods -n kepler
NAME           READY   STATUS    RESTARTS        AGE
kepler-s7jjb   1/1     Running   5 (4h29m ago)   4h31m

```
I have also confirmed this by taking metric data output using ```curl localhost:9102/metrics``` command:

Output : - [kepler_arm_output.txt](https://github.com/ossdev07/ARMFAAS/files/11543262/kepler_arm_output.txt)

I will share the changes if you plan to release the arm64 docker image.

odidev, 2023-05-30:
Hi Team,

Adding a gentle reminder to the above [comment](https://github.com/sustainable-computing-io/kepler/issues/482#issuecomment-1560917537).

rootfs, 2023-05-30:
what is your arm platform spec @odidev

odidev, 2023-05-31:
Hi @rootfs

I have used an AWS ec2 instance. Below are the platform specifications: - 

- Machine type: - t4g.medium
- Architecture: - ARM64 
- Processor: - Graviton2
- Operating System: - Ubuntu 22.04

rootfs, 2023-05-31:
Unfortunately, Graviton doesn't publish any power consumption API spec. Kepler cannot get power reading from this architecture.

odidev, 2023-06-02:
I have tried to run Kepler on an Ampere-based processor on GCP. Below are the platform specifications: - 

- Machine type: - t2a-standard-2
- Architecture: - ARM64
- Processor: - [Ampere Altra Arm processor](https://github.com/sustainable-computing-io/kepler/issues/482#t2a_machines)
- Operating System: - Ubuntu 22.04
- 
But, I am still not able to fetch the power consumption data. Below are the metrics data of Kepler.

- [kepler-t2a-standard-2-metrics.txt](https://github.com/sustainable-computing-io/kepler/files/11634442/kepler-t2a-standard-2-metrics.txt)

Please let me know the supported arm64 configurations for Kepler, where I can get the power consumption data.

rootfs, 2023-06-02:
There are two power models in Kepler: on bare metal machines, Kepler reads from the HW sensors to get the power. Bare metal Ampere is supported. On VMs (such as the GCP instance you are using), Kepler uses ML models to estimate power. At the moment, we have not trained Ampere based power models yet. Existing models are all x86 based.

odidev, 2023-06-05:
Is their any plan to release the Kepler Arm64 docker images which supports the graviton2 or Ampere based processors for VM based deployment?

closed 2023-08-28 by stale[bot]

Maarham, 2024-06-12:
Hi! I wanted to follow up on this thread. Has there been any development in Kepler running on AWS Graviton3? I see ARM64 compatibility in the MakeFile within the Github. Was wondering if some developments have been made. Thanks!