sustainable-computing-io/kepler #487: Incorrect per-Pod power usage for test benchmark

dagrayvid, 2023-01-06:
**Describe the bug**
I'm running the stress-ng benchmark for a 10 minute runtime, configured to stress ~50% of the CPUs available in my system, and seeing <5 watts attributed to the workload Pod from the grafana dashboard. 

Under load vs at idle I can see a delta of 200 Watts from the iDRAC. From prometheus I see a similar delta for irate(kepler_node_package_joules_total[1m]).  As a sanity check, I also verified that there is >50% cpu usage on the node, and and 60 cores are used according to `pod:container_cpu_usage:sum` in Prometheus.

This is running on one worker node in an OpenShift cluster with 3 control-plane and 2 worker-nodes, all with Intel Icelake CPUs.

**To Reproduce**
I'm running stress-ng with the [benchmark-operator](https://github.com/cloud-bulldozer/benchmark-operator). Under the hood, this runs stress-ng in a container with this job file:
```
run parallel  # run stressors sequentially
verbose	      # verbose output
metrics-brief # show metrics at end of each run
timeout 480   # stop each stressor after 480 seconds
cpu 64        # cpu stressors
cpu-load 100  # percentage to which the cpu gets loaded 
```

**Expected behavior**
>100 Watts attributed to the stress-ng workload Pod.

**Screenshots**
Kepler dashboard:
[image]

Kepler dashboard all namespaces:
[image]

iDRAC:
[image]

Total Node power usage according to kepler per-node metrics:
[image]

k8s hardware resource usage during run:
[image]

/cc @rootfs @sunya-ch

rootfs, 2023-01-09:
thank you @dagrayvid for the issue!

rootfs, 2023-01-09:
This *might* be resolved by a better power model, opened #490 for investigation

sunya-ch, 2023-01-10:
Thank you for reporting the issue. 
Have you tried updating the Kepler with latest PR https://github.com/sustainable-computing-io/kepler/pull/486 with the new model in https://github.com/sustainable-computing-io/kepler-model-server/pull/64?
I cannot see difference in RAPL power when setting stressors=2, cpu_percentage=50.
Thus, I tried reproducing by stressing the test node with stressors =32, cpu_percentage=100 (my vm has 32 vcores).
I have virtual machine (vm) running on top of the bare metal (bm). The workload was executed and reported by power model on vm while the RAPL power was exported on bm.

I found about 50 watts difference during the stress observed in bm-level which corresponding to reported value on vm-level.

[image]

[image]

rootfs, 2023-01-10:
@sunya-ch thank you for the validation! On the BM setup, do you run the estimator sidecar?

sunya-ch, 2023-01-10:
> @sunya-ch thank you for the validation! On the BM setup, do you run the estimator sidecar?

No estimator sidecar on BM. I run kepler with the side car only on VM because I just use BM for exporting the RAPL package power not for container-level power (ratio approach is applied).

dagrayvid, 2023-01-10:
``Thanks @sunya-ch, @rootfs. With kepler redeployed with the latest image, I am still seeing the same behaviour.
Version from kepler exporter:
`I0110 22:07:31.385345       1 exporter.go:149] Kepler running on version: 6074fce`

Node metrics show a diff of ~150 Watts from RAPL:
[image]

Container PKG joules aren't going above 5 Watts for the stress-ng pod (yellow). 
[image]

Could I be doing something wrong in deploying the exporter? Here is the kepler-exporter daemonset:
```
$ oc get daemonset/kepler-exporter -n kepler  -o yaml
apiVersion: apps/v1
kind: DaemonSet
metadata:
  ...
spec:
  revisionHistoryLimit: 10
  selector:
    matchLabels:
      app.kubernetes.io/component: exporter
      app.kubernetes.io/name: kepler-exporter
      sustainable-computing.io/app: kepler
  template:
    metadata:
      creationTimestamp: null
      labels:
        app.kubernetes.io/component: exporter
        app.kubernetes.io/name: kepler-exporter
        sustainable-computing.io/app: kepler
    spec:
      affinity:
        nodeAffinity:
          requiredDuringSchedulingIgnoredDuringExecution:
            nodeSelectorTerms:
            - matchExpressions:
              - key: feature.node.kubernetes.io/cpu-cpuid.HYPERVISOR
                operator: DoesNotExist
      containers:
      - args:
        - /usr/bin/kepler -v=5
        command:
        - /bin/sh
        - -c
        env:
        - name: NODE_NAME
          valueFrom:
            fieldRef:
              apiVersion: v1
              fieldPath: spec.nodeName
        image: quay.io/sustainable_computing_io/kepler:latest
        imagePullPolicy: Always
        livenessProbe:
          failureThreshold: 5
          httpGet:
            path: /healthz
            port: 9102
            scheme: HTTP
          initialDelaySeconds: 10
          periodSeconds: 60
          successThreshold: 1
          timeoutSeconds: 10
        name: kepler-exporter
        ports:
        - containerPort: 9102
          hostPort: 9102
          name: http
          protocol: TCP
        resources:
          requests:
            cpu: 100m
            memory: 400Mi
        securityContext:
          privileged: true
        terminationMessagePath: /dev/termination-log
        terminationMessagePolicy: File
        volumeMounts:
        - mountPath: /usr/src/kernels
          name: kernel-src
        - mountPath: /sys/kernel/debug
          name: kernel-debug
        - mountPath: /lib/modules
          name: lib-modules
        - mountPath: /sys
          name: tracing
        - mountPath: /proc
          name: proc
        - mountPath: /etc/config
          name: cfm
          readOnly: true
      dnsPolicy: ClusterFirstWithHostNet
      hostNetwork: true
      restartPolicy: Always
      schedulerName: default-scheduler
      securityContext: {}
      serviceAccount: kepler-sa
      serviceAccountName: kepler-sa
      terminationGracePeriodSeconds: 30
      tolerations:
      - effect: NoSchedule
        key: node-role.kubernetes.io/master
      volumes:
      - hostPath:
          path: /sys/kernel/debug
          type: Directory
        name: kernel-debug
      - hostPath:
          path: /usr/src/kernels
          type: Directory
        name: kernel-src
      - hostPath:
          path: /lib/modules
          type: Directory
        name: lib-modules
      - hostPath:
          path: /sys
          type: Directory
        name: tracing
      - hostPath:
          path: /proc
          type: Directory
        name: proc
      - configMap:
          defaultMode: 420
          name: kepler-cfm
        name: cfm
  updateStrategy:
    rollingUpdate:
      maxSurge: 0
      maxUnavailable: 1
    type: RollingUpdate
status:
  currentNumberScheduled: 5
  desiredNumberScheduled: 5
  numberAvailable: 5
  numberMisscheduled: 0
  numberReady: 5
  observedGeneration: 1
  updatedNumberScheduled: 5
```

sunya-ch, 2023-01-11:
> ``Thanks @sunya-ch, @rootfs. With kepler redeployed with the latest image, I am still seeing the same behaviour. Version from kepler exporter: `I0110 22:07:31.385345 1 exporter.go:149] Kepler [...]

Thank you for confirming. 
I just notice that the RAPL power was also exported. 
Is it from the same level of Kepler that was exporting the stress-ng power?  In that case, the ratio power model will be applied. 
reference: https://sustainable-computing.io/design/power_model/ 

Seeing the stats in benchmark-operator namespace, the benchmark-controller-manager pod seems to take the most of resource usage and get the most ratio of power usage. You may confirm this by showing the cpu per pod in that namespace.

In my understanding, the ratio power model is now under investigation to handle this kind of unfair power division issue.

dagrayvid, 2023-01-11:
Thanks @sunya-ch,

> Is it from the same level of Kepler that was exporting the stress-ng power?

Yes, this is a baremetal OCP cluster, and Kepler is running in a Pod but there is no virtualization involved.

> Seeing the stats in benchmark-operator namespace, the benchmark-controller-manager pod seems to take the most of resource usage and get the most ratio of power usage. You may confirm this by showing the cpu per pod in that namespace.

During the run the stress-ng Pod has the highest CPU usage in the namespace.

[image]

marceloamaral, 2023-01-11:
Thanks @dagrayvid for reporting this. I have a couple of comments here.

> Container PKG joules aren't going above 5 Watts for the stress-ng pod (yellow).

The "ratio" power model use the CPU Instruction to divide the power consumption across container + system_processes. Where the system processes are the OS + any other background processes that are not into k8s containers.
More specifically, the stress-ng power ratio will be `(stress-ng-cpu-instruction/total-cpu-instruction) * RAPL power` (note that we also consider the DRAM and other components but I will not detail it here).

So, if the stress-ng power is low, there are two possible problems: 1) the cpu instruction is not been collected, 2) there are other more intensive processes running in the node.

I suspect that the problem might be related to the CPU instructions. We export the hardware counters per container and system_processes, can you use the following prometheus query?
```
rate(kepler_container_cpu_instr_total{}[1m])
```

Also, could you please provide the kepler logs?

rootfs, 2023-01-11:
@dagrayvid can you test a new kepler image using #494? quay.io/sustainable_computing_io/kepler:pr-494

dagrayvid, 2023-01-12:
> The "ratio" power model use the CPU Instruction to divide the power consumption across container + system_processes. Where the system processes are the OS + any other background processes that are not into k8s containers.
More specifically, the stress-ng power ratio will be (stress-ng-cpu-instruction/total-cpu-instruction) * RAPL power (note that we also consider the DRAM and other components but I will not detail it here).

Thanks @marceloamaral for the explanation. It makes sense.

> So, if the stress-ng power is low, there are two possible problems: 1) the cpu instruction is not been collected, 2) there are other more intensive processes running in the node.
I suspect that the problem might be related to the CPU instructions. 

Agreed.

> We export the hardware counters per container and system_processes, can you use the following prometheus query? rate(kepler_container_cpu_instr_total[1m])
I'm not able to find this metric in prometheus.

> Also, could you please provide the kepler logs?
I attached the logs for the kepler-exporter running on the node with the stress-ng workload Pod.

@rootfs , I tried the pr-494 image, however I am still seeing similar behavior. It's very strange that the benchmark-operator Pod shows such a large Power usage, whereas the CPU usage is (and should be) much lower for that Pod.
[image]
[image]

rootfs, 2023-01-12:
I tested it on my laptop with the stressng workload.

With image pr-494, the stressng power went as high as 20W
[image]

With the latest image (aka no PR 494), the stressng power is no more than 2W.
[image]

PR #494 (2023-01-20): ebpf: fix cpu time and HW stats for running processes
fix #487