sustainable-computing-io/kepler #490: Add uops counter to baremetal power model

rootfs, 2023-01-09:
**Is your feature request related to a problem? Please describe.**
Current BM based power model uses cpu instruction/cycle/cache miss. They don't account for heterogeneous workload where cpu instructions vary between workload.

For instance, when running `stress-ng`, the instructions/cycle vary quite a bit under different cpu methods:
```console
# perf stat  -e instructions,uops_issued.any,uops_executed.thread,cycles stress-ng --cpu 1 --cpu-method int64 -t 10s
stress-ng: info:  [2801582] setting to a 10 second run per stressor
stress-ng: info:  [2801582] dispatching hogs: 1 cpu
stress-ng: info:  [2801582] successful run completed in 10.00s

 Performance counter stats for 'stress-ng --cpu 1 --cpu-method int64 -t 10s':

   100,867,201,591      instructions              #    2.60  insn per cycle           (49.99%)
   106,951,526,496      uops_issued.any                                               (50.01%)
   113,264,283,911      uops_executed.thread                                          (50.02%)
    38,862,802,583      cycles                                                        (49.99%)

      10.008234782 seconds time elapsed

       9.944586000 seconds user
       0.015193000 seconds sys

# perf stat  -e instructions,uops_issued.any,uops_executed.thread,cycles stress-ng --cpu 1 --cpu-method all -t 10s
stress-ng: info:  [2801898] setting to a 10 second run per stressor
stress-ng: info:  [2801898] dispatching hogs: 1 cpu
stress-ng: info:  [2801898] successful run completed in 10.00s

 Performance counter stats for 'stress-ng --cpu 1 --cpu-method all -t 10s':

    66,283,452,095      instructions              #    1.72  insn per cycle           (50.01%)
    78,812,841,605      uops_issued.any                                               (49.99%)
    73,867,605,976      uops_executed.thread                                          (50.00%)
    38,579,937,422      cycles                                                        (50.01%)

      10.008057024 seconds time elapsed

       9.937062000 seconds user
       0.010269000 seconds sys
```
 
**Describe the solution you'd like**
One [prior](https://lca.ece.utexas.edu/pubs/bircher-TC2012.pdf) study uses uops to resolve the workload variation and delivered excellent accuracy.

As seen in the above test results, `uops/instruction` does have some indication of the type of workload 

**Additional context**
Add any other context or screenshots about the feature request here.

jichenjc, 2023-01-12:
just for education, I searched `uops counter` but didn't find too much info, so we want to add this counter 
and maybe use to do something we did for other counters we have now?

ValZapod, 2023-02-05:
>searched uops counter but didn't find too much info

These are uops mnemonics. Bigcore is just what Intel calls what we call CPU, because main core in actually on the motherboard, in the PCH: https://github.com/chip-red-pill/uCodeDisasm/blob/master/misc/bigcore_opcodes.txt

rootfs, 2023-02-05:
Thank you @ValZapod for the pointer

marceloamaral, 2023-03-08:
The problem is that not all CPU models expose `uops`, which I do agree that is more precise than instructions. Since different instructions can execute a different number of `uops`.
As a solution, we could verify if the system supports uops and then utilize either uops or instructions, as they are interchangeable we don't need both.

However, it is very important to make it very clear to the user, as this will generate different power models.

closed 2023-05-24 by stale[bot]