sustainable-computing-io/kepler #495: Export kepler_container_cgroup/bpf_(resource usage metric) for model training

sunya-ch, 2023-01-12:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

According to the metric list [here](https://sustainable-computing.io/design/metrics/#kepler-metrics-overview), there are only hardware counter metrics exported. 
We also need the other set of metrics to build the model that is available on the system that has no hardware counter.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

We can name it corresponding to what we define in the config/types.go and label with the source.
 
https://github.com/sustainable-computing-io/kepler/blob/20ccfe414b0329a6b14bb0c907c5be27b3c2aa71/pkg/config/types.go#L26-L45

Candidates:

- kepler_container_cpu_usage_us_total{source="cgroupv2"}
- kepler_container_memory_usage_bytes_total{source="cgroupv2"}
- kepler_container_system_cpu_usage_us_total{source="cgroupv2"}
- kepler_container_user_cpu_usage_us_total{source="cgroupv2"}
- kepler_container_cpu_usage_seconds_total{source="kubelet"}
- kepler_container_memory_working_set_bytes_total{source="kubelet"}
- kepler_container_cpu_time_(us/ns/seconds)_total{source="bpf"}

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

jichenjc, 2023-01-12:
so you just want to expose all those metrics listed above ? if so I can give a try if no one else is working on it

sunya-ch, 2023-01-17:
> so you just want to expose all those metrics listed above ? if so I can give a try if no one else is working on it

Yes, I would like to have all metrics available if it is enabled. They would be enabled only in the system that can and want to contribute on building the model in an online manner. 

I would very appreciate your help :D

sunya-ch, 2023-01-25:
@jichenjc  will you also work on cgroup metrics? Otherwise, I will work on that. I need those metrics for refactoring the kepler-model-server training pipeline?

jichenjc, 2023-01-30:
>>.will you also work on cgroup metrics? Otherwise, I will work on that. I need those metrics for refactoring the kepler-model-server training pipeline?

yes, please ,I just return back from holiday so please submit PR thus I won't block your progress.. thanks~

KaiyiLiu1234, 2023-02-24:
Could you change it so that kepler_container_cpu_cpu_time_us is instead kepler_container_cpu_time_total (remove the extra cpu)?

sunya-ch, 2023-02-24:
> Could you change it so that kepler_container_cpu_cpu_time_us is instead kepler_container_cpu_time_total (remove the extra cpu)?

update to `kepler_container_bpf_cpu_time_us_total`. Please review this PR https://github.com/sustainable-computing-io/kepler/pull/547.

closed 2023-03-07 by sunya-ch