sustainable-computing-io/kepler #497: idea: present Kepler on eCHO Livestream

lizrice, 2023-01-13:
I'm one of the hosts of a weekly livestream about eBPF technology, called [eCHO](https://youtube.com/playlist?list=PLDg_GiBbAx-mY3VFLPbLHcxo6wUjejAOC). I think Kepler would make a really great topic for an episode. Is there someone who would like to come and show it to our audience?

rootfs, 2023-01-13:
Thank you for the invitation @lizrice! I have watched quite some episodes. I am excited to join your show. It will be great news for Kepler and eBPF community on the show.

lizrice, 2023-01-16:
Brilliant! I reached out to you with a message on LinkedIn so we can co-ordinate a date

lizrice, 2023-03-18:
Done in [Episode 80](https://www.youtube.com/live/xzfTU_Wa7rU?feature=share)

closed 2023-03-18 by lizrice