sustainable-computing-io/kepler #499: CI: failed to build image

rootfs, 2023-01-19:
**Describe the bug**
Recent CI failure in image_build.yml
https://github.com/sustainable-computing-io/kepler/actions/runs/3954494189/jobs/6771887553
```
Download action repository 'actions/checkout@v3' (SHA:ac593985615ec2ede58e132d2e21d2b1cbd6127c)
Download action repository 'actions/setup-go@main' (SHA:21459d0b7b1d63741429b748885bf5a4974593b4)
Warning: Failed to download action 'https://api.github.com/repos/actions/setup-go/tarball/21459d0b7b1d63741429b748885bf5a4974593b4'. Error: Error while copying content to a stream.
Warning: Back off 22.187 seconds before retry.
Warning: Failed to download action 'https://api.github.com/repos/actions/setup-go/tarball/21459d0b7b1d63741429b748885bf5a4974593b4'. Error: Response status code does not indicate success: 503 (Service Unavailable).
Warning: Back off 16.[33](https://github.com/sustainable-computing-io/kepler/actions/runs/3954494189/jobs/6771887553#step:1:37)3 seconds before retry.
Error: The request was canceled due to the configured HttpClient.Timeout of 100 seconds elapsing.
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

rootfs, 2023-01-19:
github outage issue, not related to the recent code change
[image]

closed 2023-01-19 by rootfs

SamYuan1990, 2023-01-19:
just trigger the job.