sustainable-computing-io/kepler #503: kepler MVC architecture

rootfs, 2023-01-24:
In a classic MVC model, M for model, V for view, C for control.
Considering as our kepler endpoint is a website, 
- At view level:
We need/are going to support both prometheus format and OTeL format, we can support different format by use different view as output.

- At model level:
We can define kepler core model.

- At Control level:
Considering with multiple CPU arch or hardware(GPU) or data source(ex. cgroup), we can have a general control interface and each type of hardware can has it's own control logic to update data.

rootfs, 2023-01-24:
discussion in 0.5 release cycle.
1) investigate the architectural overhead of metric collecting and rendering.
2) investigate the otel overhead and architectural impact

In standalone mode, exporter needs to be extracted. It could be an organic step in this process.

SamYuan1990, 2023-02-05:
in my point of view, if we don't have count to sum convert, we can just replace our internal impl to Otel?
ref to Otel sample https://github.com/open-telemetry/opentelemetry-go/tree/main/example/prometheus

SamYuan1990, 2023-02-05:
@rootfs , may I know more about Otel use case if we want to use kepler in edge node?
as sample above still a web endpoint with get method.

rootfs, 2023-02-06:
@SamYuan1990 let's first investigate the overhead of constructing metrics. There will be lots of optimization in place before we can support telemetry.

closed 2023-05-24 by stale[bot]