sustainable-computing-io/kepler #505: Add benchmark testing.

rootfs, 2023-01-24:
Add benchmark test cases, performance (memory issue), should we fix the memory issue or keep investigating? 

0.5 release.

rootfs, 2023-01-24:
@marceloamaral local benchmark tool to share

jichenjc, 2023-01-31:
duplicate to #506?

SamYuan1990, 2023-02-05:
> duplicate to #506?

no, @jichenjc there two different level of benchmark testing.
1st is code level as go test bench 
2nd is integration level as #487

jichenjc, 2023-02-05:
@SamYuan1990 ok, make sense to me :)

SamYuan1990, 2023-03-31:
@rootfs , @marceloamaral , @sunya-ch 
may I know for integration test(or say keeping create pod on k8s cluster to test kepler)
are we able to just deploy https://github.com/cloud-bulldozer/benchmark-operator#workloads-status on the same cluster?
if in this approach works, it seems not need to change kepler code or just have a POC or document as reference.
ref #487

SamYuan1990, 2023-05-17:
@marceloamaral , @sunya-ch 
I am not sure if we can use [CPE](https://github.com/IBM/cpe-operator) to keep generate pods for integration test with for example 1000 pods as workload?

marceloamaral, 2023-05-18:
We can use CPE and/or have a more simpler approach inside Kepler.
I am working on that.

sunya-ch, 2023-06-22:
> @marceloamaral , @sunya-ch
> I am not sure if we can use [CPE](https://github.com/IBM/cpe-operator) to keep generate pods for integration test with for example 1000 pods as workload?

You may think CPE is only the wrapper of benchmark operator. It generates a k8s job list and then create a job sequentially to run when the previous has completed. In parallel, it depends on the scalability of the API server whether it can generate a thousand of resource at the same time.

sunya-ch, 2023-08-18:
I started implementing an integration in Kepler-Model-Server project. 
I think we can reuse it for the Kepler integration test or move it [Kepler-action](https://github.com/sustainable-computing-io/kepler-action). 
https://github.com/sustainable-computing-io/kepler-model-server/tree/main/tests#integration-test

sunya-ch, 2023-08-21:
@SamYuan1990 
It is test case for integrate kepler with model server and should be covered by the kepler operator too. 
The action should set image as a variable. 

The cases of use include 
1. Kepler-model-server: Build developing kepler-model-sever image and deploy with latest image of kepler
2. Kepler (exporter): Build developing kepler image and deploy with latest kepler-model-server image
3. Kepler-operator: Deploy with latest image of kepler and kepler-model-server

rootfs, 2023-09-21:
@mcalman can you take a look and see if your scalability test be able to used in the CI?

SamYuan1990, 2023-10-01:
> @SamYuan1990 It is test case for integrate kepler with model server and should be covered by the kepler operator too. The action should set image as a variable.
>
> The cases of use include
>
> 1. [...]

if so.... I am a bit confusing for where the action belongs to....
for example, kepler exporter's test should belongs to kepler exporter's repo. which means, it's not going to build latest kepler-model-server image?

sunya-ch, 2023-10-02:
> if so.... I am a bit confusing for where the action belongs to.... for example, kepler exporter's test should belongs to kepler exporter's repo. which means, it's not going to build latest [...]

Yes, I think exporter test should not be failed if the unstable version of kepler-model-server image is failed and vice versa.

closed 2023-12-08 by stale[bot]