sustainable-computing-io/kepler #506: Further enhancement on kepler test coverage

SamYuan1990, 2023-01-24:
Currently for kepler test case. By code structure limitation, most of code we just run the test and ensure the code will not panic, and crash kepler service.
Further enhancement on kepler test coverage is needed as improvement of code quality.

Stretch in 0.5 release: performance and scalability (100+ small pods and investigate resource and performance impact on Kepler) and stress tests, can be done offline (not enabled in CI)

jichenjc, 2023-01-31:
>>>Stretch in 0.5 release: performance and scalability (100+ small pods and investigate resource and performance impact on Kepler) and stress tests, can be done offline (not enabled in CI)

I think we should still keep those codes in an automatic way (e.g in the CI repo), correct?
and kepler is per node daemonset, thus, 100+ small pods need to be considered scatter into just 1 node or 100 node?

jichenjc, 2023-02-03:
@SamYuan1990 we may follow  https://github.com/sustainable-computing-io/kepler/issues/487

SamYuan1990, 2023-02-04:
thanks I will check it later.

SamYuan1990, 2023-02-05:
@rootfs , @marceloamaral I suppose we need to add benchmark testing at functional level for kepler's data from each source. for example, cgroup, ebpf as bcc, kebulete etc...
could you please help list all of those here?
I found one and it becomes a sample PR as:
https://github.com/sustainable-computing-io/kepler/pull/521

SamYuan1990, 2023-02-11:
I suppose we need add benchmark for major branch under https://github.com/sustainable-computing-io/kepler/tree/main/pkg/collector for our collectors.

SamYuan1990, 2023-02-12:
```
updateBPFMetrics
updateCgroupMetrics
updateKubeletMetrics
updateNodeEnergyMetrics
```
@rootfs , I suppose we need to add benchmark testing for functions above?

jichenjc, 2023-02-13:
updateCgroupMetrics/updateKubeletMetrics

to confirm my understanding, seems those function both rely on `range c.ContainersMetrics`, so guess we need test scale the container size (e.g 100, 1000) then do the perf?

SamYuan1990, 2023-02-13:
> updateCgroupMetrics/updateKubeletMetrics
> 
> to confirm my understanding, seems those function both rely on `range c.ContainersMetrics`, so guess we need test scale the container size (e.g 100, 1000) then do the perf?

most of our critical path are create or update on ContainersMetrics, hence so far we just test with size of this structure as 1, 2k, 5k, 10k size.

But I am doubt if it makes sense or not ... for some times.
As those thing mostly after IO, hence the test only covers calculate part.
So loop 10k size of array or map, is not too much effort.
I am wondering if we should considering contention?
or by current implementation, I still not cover performance bottleneck yet?

rootfs, 2023-02-13:
It is great to have benchmarks around the critical path. We have seen latency in container ID resolution, metric updating and prometheus scraping. @SamYuan1990 has written benchmarks on metrics update. It would be great to also have one on the prometheus collector

SamYuan1990, 2023-02-19:
is there any idea to mock cgroup? currently test case it's based on mock file. but, seems just single container. which means, if we are going to mock for branch testing, we should make a file with thousands of lines?

SamYuan1990, 2023-05-17:
I suppose we can close this ticket for now. as rest of benchmark at code level related to either ebpf or cgroup. which out of kepler scope. cc: @rootfs , @marceloamaral  please reopen it if needed.

closed 2023-05-17 by SamYuan1990