sustainable-computing-io/kepler #507: make cluster-up available

rootfs, 2023-01-24:
create a dedicated artifact or repo for make cluster-up and integrate with CI

rootfs, 2023-01-24:
1) simple way to clone/retrieve the artifact
2) kepler repo: make cluster-up and get the repo

SamYuan1990, 2023-01-25:
> * simple way to clone/retrieve the artifact

wait.... you mean bcc?

SamYuan1990, 2023-01-25:
@rootfs , @marceloamaral I suppose we don't need to add bcc as a part of this ticket scope. as we don't own bcc. and our bcc deb packages, well, just for ubuntu OS. which means even if we implemented  **a simple way to clone/retrieve the artifact** client still need to make their bcc libs.

SamYuan1990, 2023-01-25:
for kepler repo and kepler action repo, @rootfs , @marceloamaral `git submodule` or just clone during make script running? which way do you prefer?

rootfs, 2023-01-25:
@SamYuan1990 right, bcc is not a requirement. The cluster-up and sync are needed.

closed 2023-01-30 by rootfs