sustainable-computing-io/kepler #510: allow Kepler to run if no Pods info exist

rootfs, 2023-01-25:
**Is your feature request related to a problem? Please describe.**
Currently, kepler aggregates metrics at Pods level when the cgroup is resolved and container ID can be retrieved from kubelet. In environment without kubelet, kepler should still export metrics, just not to aggregate them at pods level. 

**Describe the solution you'd like**
Some env var to enable/disable kubelet query and allow kepler to export metrics without pod level aggregation

jichenjc, 2023-01-31:
>>>In environment without kubelet, kepler should still export metrics, just not to aggregate them at pods level.

so this is for something like docker-componse that not in k8s ecosystem but still containerized workload?

jichenjc, 2023-02-01:
um... checked the code 
https://github.com/sustainable-computing-io/kepler/blob/main/pkg/cgroup/resolve_container.go#L142
https://github.com/sustainable-computing-io/kepler/blob/main/pkg/cgroup/resolve_container.go#L326

seems we highly rely on above logic (get pods from kubelet) then resolve the containerID from pod info such as init/ephemeral/normal pod's each container info , if we remove the kubelet query, where should we obtain such info? so it's not only the pod info , but also the container info ?

marceloamaral, 2023-05-18:
@rootfs and @jichenjc since we have the process metrics can we close this issue?

closed 2023-08-28 by stale[bot]