sustainable-computing-io/kepler #515: package Kepler for standalone distribution

rootfs, 2023-02-01:
**Is your feature request related to a problem? Please describe.**
Kepler is distributed in a container for Kubernetes. However, when deployed in a standalone and non-containerized environment, a software packaging scheme, rpm or deb, is needed.

**Describe the solution you'd like**
Initially a Kepler rpm should be supported

rootfs, 2023-02-09:
Using Github Action to create the rpm and hosting the manifests there is a good start.

SamYuan1990, 2023-02-09:
personally, binary is good enough and upload binaries to release page?
https://github.com/prometheus/node_exporter/releases/tag/v1.5.0
for systemd we may able to ref https://github.com/prometheus/node_exporter/tree/master/examples/systemd

SamYuan1990, 2023-02-09:
We just have a binary for kepler as release which is good enough right?
if we provide a systemd job, do we need root privileges for `/proc` access? or we need root privileges to open network so that this systemd job able to listen on port as 9102?a
if my point of view a binary release is a minimum size of standalone. A systemd job contains more features as start, stop, auto recover, log, privileges etc.

husky-parul, 2023-02-10:
> We just have a binary for kepler as release which is good enough right? if we provide a systemd job, do we need root privileges for `/proc` access? or we need root privileges to open network so that [...]

We are starting with rpm binary. Wanted to clarify if we intend to support start/stop/log etc.

SamYuan1990, 2023-02-10:
Pre offline discussion with @rootfs
@husky-parul I am good if we can put binary in release page and rpm package on any rpm package center.

closed 2023-05-24 by stale[bot]

Stef16Robbe, 2023-12-20:
Are there any plans of supporting other packaging formats like .deb?