sustainable-computing-io/kepler #517: Kepler with K3s on Virual machine

EmilioIppoliti, 2023-02-02:
**Describe the bug**
Hi, 
I am trying to install kepler on a k3s s Virtual machine Ubuntu 20 (kernel 5.15) , however it would appear that kepler is unable to expose container metrics, when I made "curl http://127.0.0.1:9102/metrics | grep container"
I receive no metric about container 

[image]

rootfs, 2023-02-02:
@EmilioIppoliti can you check if your VM is running cgroup v2, i.e. `mount |grep cgroup`?

On the VM, kepler uses cgroup stats to estimate energy consumption. Currently only cgroup v2 is supported.

jichenjc, 2023-02-03:
yes, this looks familiar to the issue I used to meet, maybe related to cgroupv2 

and Ubuntu seems should work but I think CentOS might be a better starting point

EmilioIppoliti, 2023-02-03:
I really thank you for you answers , anyway I checked the correct version of my cgroups in my linux distro ,
I think they are mounted correctly , it could be something related to virtualisation or k3s ?
[image]

Thank you for your support

rootfs, 2023-02-03:
@EmilioIppoliti can you turn on verbosity to `v=5` in kepler daemonset and share the log? How is your k3s configured? Does it use cri-o or containerd? If you can get the containers' cgroup path, that'll help us to check if the paths are expected by kepler, e.g. when using cri-o, you can use `crictl` to inspect the pod/containers:
```console
# crictl inspect 5cad2f004420c |grep -i cgroup
          "destination": "/sys/fs/cgroup",
          "type": "cgroup",
          "source": "cgroup",
        "cgroupsPath": "kubepods-burstable-pod1a0de199_47c2_4db5_a593_0121bcd42e96.slice:crio:5cad2f004420cf8cdc11f8e4813dec9c658a85851645c14d126e518178d038ff",
            "type": "cgroup"
```

jichenjc, 2023-02-05:
and usually the startup logs can provide some key wording about settings and potential issue about metric report, so can you restart the pod and wait for around 1 min then paste startup logs in addition to above output as well?

```
kubectl rollout restart ds/kepler-exporter -n kepler
kubectl logs ds/kepler-exporter -n kepler  -c kepler-exporter
```

EmilioIppoliti, 2023-02-06:
I really Thank you for your answer ,  I set the verbosity of the log to ten . 
But I was able to get more metrics , installing kepler from source code and not from the helm chart .
Maybe the source code and the helm chart have some differences?

rootfs, 2023-02-06:
Cool! We'll update the chart 

cc @jiangphcn

jiangphcn, 2023-02-07:
thanks for report. I will work on this.

marceloamaral, 2023-05-18:
@jiangphcn any update on this?

rootfs, 2023-05-18:
New helm chart has tested on VM with k3s

closed 2023-05-18 by rootfs

EmilioIppolitiHPE, 2023-05-18:
I tried the ney helm chart and i really thank you for your support , I think that this request can be closed

rootfs, 2023-05-18:
thanks for verification