sustainable-computing-io/kepler #524: logs about node Energy is confusing

jichenjc, 2023-02-07:
```
I0207 01:18:49.668971       1 metric_collector.go:130] node energy (mJ):
        ePkg: 0 (eCore: 0 eDram: 0 eUncore: 0) eGPU: 0 eOther: 0
```

```
# HELP kepler_node_core_joules_total Aggregated RAPL value in core in joules
# TYPE kepler_node_core_joules_total counter
kepler_node_core_joules_total{instance="jitest42.novalocal",mode="dynamic",package="0",source="rapl"} 0.07
kepler_node_core_joules_total{instance="jitest42.novalocal",mode="idle",package="0",source="rapl"} 0.17
# HELP kepler_node_dram_joules_total Aggregated RAPL value in dram in joules
# TYPE kepler_node_dram_joules_total counter
kepler_node_dram_joules_total{instance="jitest42.novalocal",mode="dynamic",package="0",source="rapl"} 685.851
kepler_node_dram_joules_total{instance="jitest42.novalocal",mode="idle",package="0",source="rapl"} 13.043
# HELP kepler_node_energy_stat Several labeled node metrics
# TYPE kepler_node_energy_stat counter
kepler_node_energy_stat{cpu_architecture="Haswell",node_block_devices_used="0",node_curr_bytes_read="0",node_curr_bytes_writes="0",node_curr_cache_miss="0",node_curr_container_cpu_usage_seconds_total="0",node_curr_container_memory_working_set_bytes="0",node_curr_cpu_cycles="0",node_curr_cpu_instr="0",node_curr_cpu_time="0",node_curr_energy_in_core_joule="0",node_curr_energy_in_dram_joule="0",node_curr_energy_in_gpu_joule="0",node_curr_energy_in_other_joule="0",node_curr_energy_in_pkg_joule="0",node_curr_energy_in_uncore_joule="0",node_name="jitest42.novalocal"} 0
# HELP kepler_node_nodeInfo Labeled node information
# TYPE kepler_node_nodeInfo counter
kepler_node_nodeInfo{cpu_architecture="Haswell"} 1
# HELP kepler_node_other_host_components_joules_total Aggregated RAPL value in other components (platform - package - dram) in joules
# TYPE kepler_node_other_host_components_joules_total counter
kepler_node_other_host_components_joules_total{instance="jitest42.novalocal",mode="dynamic"} 0
kepler_node_other_host_components_joules_total{instance="jitest42.novalocal",mode="idle"} 0
# HELP kepler_node_package_energy_millijoule Aggregated RAPL value in package (socket) in milijoules (deprecated)
# TYPE kepler_node_package_energy_millijoule counter
kepler_node_package_energy_millijoule{core="1",dram="121",instance="jitest42.novalocal",pkg_id="0",uncore="0"} 1
# HELP kepler_node_package_joules_total Aggregated RAPL value in package (socket) in joules
# TYPE kepler_node_package_joules_total counter
kepler_node_package_joules_total{instance="jitest42.novalocal",mode="dynamic",package="0",source="rapl"} 0.07
kepler_node_package_joules_total{instance="jitest42.novalocal",mode="idle",package="0",source="rapl"} 0.17
# HELP kepler_node_platform_joules_total Aggregated RAPL value in platform (entire node) in joules
# TYPE kepler_node_platform_joules_total counter
kepler_node_platform_joules_total{instance="jitest42.novalocal",mode="dynamic",source="acpi"} 0
kepler_node_platform_joules_total{instance="jitest42.novalocal",mode="idle",source="acpi"} 0
# HELP kepler_node_uncore_joules_total Aggregated RAPL value in uncore in joules
# TYPE kepler_node_uncore_joules_total counter
kepler_node_uncore_joules_total{instance="jitest42.novalocal",mode="dynamic",package="0",source="rapl"} 0
kepler_node_uncore_joules_total{instance="jitest42.novalocal",mode="idle",package="0",source="rapl"} 0

```

I am using a VM seems we have multiple values but not sure why logs report all 0 though..

rootfs, 2023-02-07:
the `other` domain is from power meter, and `uncore` domain is for intel gpu, it is not a surprise to see they are `0`

jichenjc, 2023-02-08:
but `ePkg: 0 (eCore: 0 eDram: 0 eUncore: 0) ` this seems incorrect to me ..

rootfs, 2023-02-08:
The log message might come from rapl reading (which is absent on VMs), the value in the metrics are from model based estimate.

jichenjc, 2023-02-09:
ok, but previously it's not in the exact same env I am using .. 
will check more and see whether I can see any clue

marceloamaral, 2023-02-09:
One clue to investigate is: prometheus shows aggregated values and the logs are probably the delta values.

PR #531 (2023-02-10): fix: make log about node energy accurate
Fixes #524
closed 2023-02-10 by rootfs