sustainable-computing-io/kepler #530: set timeout to access external when executing loadWeightFromURL

jichenjc, 2023-02-10:
in a restricted network env, takes 5 min to init the model due to not able to access the URL
we should think about decrease the timeout of the loadURL so we can get better performance when startup 

```
I0209 23:50:57.522091       1 utils.go:62] Available I/O metrics: [bytes_read bytes_writes]
I0209 23:50:57.522204       1 model.go:85] Model Config NODE_TOTAL: {UseEstimatorSidecar:false SelectedModel: SelectFilter: InitModelURL:}
I0209 23:50:57.522253       1 lr.go:169] LR Model (AbsModelWeight): no config
I0209 23:50:57.522265       1 model.go:77] Model AbsModelWeight initiated (false)
I0209 23:50:57.522276       1 model.go:85] Model Config NODE_COMPONENTS: {UseEstimatorSidecar:false SelectedModel: SelectFilter: InitModelURL:}
I0209 23:55:18.769831       1 lr.go:162] LR Model (AbsComponentModelWeight): loadWeightFromURL(https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-server/main/tests/test_models/AbsComponentModelWeight/Full/KerasCompWeightFullPipeline/KerasCompWeightFullPipeline.json): map[core:{{17.49994468688965 map[cpu_architecture:map[Alder Lake:{0.5408945679664612} Broadwell:{17.9639892578125} Cascade Lake:{-0.49166440963745117} Coffee Lake:{0.5166589617729187} Haswell:{-0.5789095163345337} Ivy Bridge:{-0.024028241634368896} Sandy Bridge:{0.5239214301109314} Sky Lake:{0.4193417429924011}]] map[cpu_cycles:{6.85713664e+09 7.560771917192364e+18 -0.11352460086345673} cpu_instr:{3.374244864e+09 8.408530291701842e+17 -0.414739191532135} cpu_time:{192019.5 5.2761312e+08 -0.06457684189081192}]}} dram:{{17.49994468688965 map[cpu_architecture:map[Alder Lake:{-0.11559933423995972} Broadwell:{16.972564697265625} Cascade Lake:{0.5505847334861755} Coffee Lake:{-0.4564790725708008} Haswell:{-0.13912856578826904} Ivy Bridge:{-0.018331050872802734} Sandy Bridge:{-0.6695247888565063} Sky Lake:{0.29698115587234497}]] map[cache_miss:{9.329199e+06 2.2145245642752e+13 -0.4680119752883911} container_memory_working_set_bytes:{253952 1.93474854912e+11 0.6805805563926697}]}}]
```

rootfs, 2023-02-10:
@jichenjc good findings! Would you like to add a makefile target to download the models and add them to kepler image?

jichenjc, 2023-02-10:
I think we do put stuffs through docker build process , so curious what's the cause , still checking

rootfs, 2023-02-10:
the models are download upon start
https://github.com/sustainable-computing-io/kepler/blob/main/pkg/model/container_power.go#L35

jichenjc, 2023-02-10:
ok, I thought we put something in the docker file 
https://github.com/sustainable-computing-io/kepler/blob/main/build/Dockerfile#L35

but seems those are different... let me see how we can optimize

jichenjc, 2023-02-10:
@sunya-ch I checked the code and found we support a configmap so that we can overwrite the URL 
https://github.com/sustainable-computing-io/kepler/blob/main/pkg/model/container_power.go#L42

so it should be fine we have a logic like
=> first check overwrite URL 
=> check the file on local (which I will add through docker build)
=> at last check the default URL 

does this sounds good to u?thanks

sunya-ch, 2023-02-10:
> @sunya-ch I checked the code and found we support a configmap so that we can overwrite the URL https://github.com/sustainable-computing-io/kepler/blob/main/pkg/model/container_power.go#L42
>
> so it [...]

Sure. I think it’s also good to add local model too. I supposed it is just a small json file, right? Also, I am not sure do we still need the previous one in Dockerfile (normalized_cpu_arch, power_data, power_model). Are we planning to use that? If not, we may remove it from the image.

jichenjc, 2023-02-10:
looks like normalized_cpu_arch.csv are still used to get node info 

the other 2 seems not used any more .. I can do a separate PR to remove power_data, power_model if everyone agree

jichenjc, 2023-02-10:
https://github.com/sustainable-computing-io/kepler/pull/532 proposed

PR #544 (2023-02-21): fix: put json file into container image to avoid URL access
Fixes: #530
closed 2023-02-21 by rootfs