sustainable-computing-io/kepler #539: How to reduce the sampling rate?

andersonandrei, 2023-02-14:
Hello everyone,

I have worked with Serverless platforms on top of Kubernetes. More specifically, OpenWhisk.
There I execute small functions with fast execution - at most 5 minutes, but about 1 minute in general.
I want to extract their energy metrics with Kepler, but it seems that the sampling rate by default is 15s. I want to reduce it to 5s or even 1s.
Can you help me, please?

Thank you very much!

rootfs, 2023-02-14:
@andersonandrei there are two places that set the sampling interval: the `ServiceMonitor` and kepler.

The default Prometheus scraping interval is `3s` in the [default `ServiceMonitor` manifest](https://github.com/sustainable-computing-io/kepler/blob/main/manifests/config/exporter/prometheus_service_monitor.yaml#L10)

The kepler sampling interval is [hardcoded](https://github.com/sustainable-computing-io/kepler/blob/main/pkg/manager/manager.go#L27) (you are welcome to make a PR to make it configurable) is 3s.

There might be some latency from Prometheus or Grafana dashboard. If you can help us figure that out, that'll be great.

andersonandrei, 2023-03-01:
Hi @rootfs , thank you very much for your answer.

I will try to work with 3s of sampling intervals for a while and then I will try to modify the code to be 1s.

Besides, I have not been able to set it to 3s yet, even using the interval of 3s as you exemplified with the [Service Monitor ](https://github.com/sustainable-computing-io/kepler/blob/main/manifests/config/exporter/prometheus_service_monitor.yaml#L10). The sampling interval that I get, through Prometheus, after deploying Kepler, is 14s.
Can you show me an example of how to set up Prometheus for that, please?

Thanks again!

marceloamaral, 2023-05-18:
@andersonandrei were you able to fix your problem?
Can we close this issue?

closed 2023-08-28 by stale[bot]