sustainable-computing-io/kepler #542: Test coverage is less than 50% for now, call for test case contribution.

SamYuan1990, 2023-02-18:
as subject, our test coverage is less than 50% for now, call for test case contribution for packages.

SamYuan1990, 2023-02-18:
https://github.com/sustainable-computing-io/kepler/actions/runs/4195087429/jobs/7274123870#step:6:24 ref here
```
--- PASS: TestBpfassets (0.01s)
PASS
coverage: 14.7% of statements

--- PASS: TestPodLoader (0.03s)
PASS
coverage: 55.8% of statements

--- PASS: TestCollector (0.06s)
PASS
coverage: 49.5% of statements

--- PASS: TestPodLoader (0.01s)
coverage: 31.6% of statements

--- PASS: TestPodLoader (0.01s)
PASS
coverage: 58.3% of statements

--- PASS: TestPodLoader (0.00s)
PASS
coverage: 53.0% of statements

--- PASS: TestPodLoader (0.00s)
PASS
coverage: 53.0% of statements

--- PASS: TestModels (0.57s)
coverage: 57.5% of statements

--- PASS: TestLocalEstimator (0.20s)
coverage: 69.5% of statements

--- PASS: TestSidecar (0.01s)
PASS
coverage: 70.6% of statements

--- PASS: TestPower (0.01s)
PASS
coverage: [no statements]

--- PASS: TestUtils (0.01s)
PASS
coverage: 51.5% of statements

--- PASS: TestVersion (0.00s)
PASS
coverage: [no statements]
```

rootfs, 2023-02-19:
```
--- PASS: TestBpfassets (0.01s)
PASS
coverage: 14.7% of statements
```

This could get improved if we exclude [perf_event_bindata.go](https://github.com/sustainable-computing-io/kepler/blob/main/pkg/bpfassets/perf_event_bindata.go) from coverage.

SamYuan1990, 2023-03-28:
via command `go list ./... | grep pkg | grep -v bpfassets` test scope will be
```
github.com/sustainable-computing-io/kepler/pkg/cgroup
github.com/sustainable-computing-io/kepler/pkg/collector
github.com/sustainable-computing-io/kepler/pkg/collector/metric
github.com/sustainable-computing-io/kepler/pkg/config
github.com/sustainable-computing-io/kepler/pkg/kubelet
github.com/sustainable-computing-io/kepler/pkg/manager
github.com/sustainable-computing-io/kepler/pkg/model
github.com/sustainable-computing-io/kepler/pkg/model/estimator/local
github.com/sustainable-computing-io/kepler/pkg/model/estimator/sidecar
github.com/sustainable-computing-io/kepler/pkg/model/types
github.com/sustainable-computing-io/kepler/pkg/power
github.com/sustainable-computing-io/kepler/pkg/power/accelerator
github.com/sustainable-computing-io/kepler/pkg/power/accelerator/source
github.com/sustainable-computing-io/kepler/pkg/power/acpi
github.com/sustainable-computing-io/kepler/pkg/power/components
github.com/sustainable-computing-io/kepler/pkg/power/components/source
github.com/sustainable-computing-io/kepler/pkg/utils
github.com/sustainable-computing-io/kepler/pkg/version
```