sustainable-computing-io/kepler #545: RHEL9.1 get failed to attach kprobe error

jichenjc, 2023-02-22:
using RHEl9.1 (though s390x but I think it's a general issue??)

```

]# _output/bin/linux_s390x/kepler
I0222 00:28:01.665839   99973 gpu_nvml.go:45] could not init nvml: <nil>
Failed to init nvml: could not init nvml: <nil>, using dummy source to obtain gpu power
I0222 00:28:01.716453   99973 exporter.go:150] Kepler running on version: v0.4-80-ge855d28-dirty
I0222 00:28:01.716564   99973 config.go:148] using gCgroup ID in the BPF program: true
I0222 00:28:01.716641   99973 config.go:149] kernel version: 5.14
I0222 00:28:01.716692   99973 config.go:167] EnabledGPU: false
I0222 00:28:01.719580   99973 slice_handler.go:179] Not able to find any valid .scope file in /sys/fs/cgroup/system.slice, this likely cause all cgroup metrics to be 0
I0222 00:28:01.719826   99973 acpi.go:77] Could not find any ACPI power meter path. Is it a VM?
cannot attach kprobe, probe entry may not exist =====> this seems not correct .. 
perf_event_open: No such file or directory

https://github.com/iovisor/bcc/blob/master/src/cc/libbpf.c#L1171

[root@gustily1 kepler]# cat /etc/os-release
NAME="Red Hat Enterprise Linux"
VERSION="9.1 (Plow)"
ID="rhel"
ID_LIKE="fedora"
VERSION_ID="9.1"
PLATFORM_ID="platform:el9"
PRETTY_NAME="Red Hat Enterprise Linux 9.1 (Plow)"
ANSI_COLOR="0;31"
LOGO="fedora-logo-icon"
CPE_NAME="cpe:/o:redhat:enterprise_linux:9::baseos"
HOME_URL="https://www.redhat.com/"
DOCUMENTATION_URL="https://access.redhat.com/documentation/red_hat_enterprise_linux/9/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"

REDHAT_BUGZILLA_PRODUCT="Red Hat Enterprise Linux 9"
REDHAT_BUGZILLA_PRODUCT_VERSION=9.1
REDHAT_SUPPORT_PRODUCT="Red Hat Enterprise Linux"
REDHAT_SUPPORT_PRODUCT_VERSION="9.1"

[root@gustily1 kepler]# uname -a
Linux gustily1.fyre.ibm.com 5.14.0-70.30.1.el9_0.s390x #1 SMP Fri Oct 14 10:28:08 EDT 2022 s390x s390x s390x GNU/Linux

```

I commented https://github.com/sustainable-computing-io/kepler/blob/e855d28b0f50a20c159382b781efcfa96807d694/pkg/bpfassets/attacher/bcc_attacher.go#L69-L79

and the error is gone, so is something new in the new linux kernel.......

rootfs, 2023-02-22:
```
cannot attach kprobe, probe entry may not exist
```

@jichenjc can you run `grep finish_task_switch /proc/kallsyms` and share the output? Here is mine:

```console
# grep finish_task_switch /proc/kallsyms 
ffffffffb8d1f730 t finish_task_switch
```

jichenjc, 2023-02-23:
this is my env's setting

```
[root@gustily1 ~]# grep finish_task_switch /proc/kallsyms
000000005a266660 t finish_task_switch.isra.0
[root@gustily1 ~]# gcc version
/usr/bin/ld: cannot find version: No such file or directory
collect2: error: ld returned 1 exit status
[root@gustily1 ~]# gcc --version
gcc (GCC) 11.3.1 20220421 (Red Hat 11.3.1-2)
Copyright (C) 2021 Free Software Foundation, Inc.

```

jichenjc, 2023-02-23:
ok, I think the issue is clear, we do have this logic already

I add such log trace
```
        fmt.Println("--------1")
        err = m.AttachKprobe("finish_task_switch", ftswitch, -1)
        if err != nil {
                fmt.Println("--------2")
                err = m.AttachKprobe("finish_task_switch.isra.0", ftswitch, -1)
                if err != nil {
                        return nil, fmt.Errorf("failed to attach finish_task_switch: %s", err)
                }
                fmt.Println("--------3")
        }

```

and got this 
```
--------1
cannot attach kprobe, probe entry may not exist
--------2
--------3

```

so we already handle such issue thus `cannot attach kprobe, probe entry may not exist` can be safely ignored

closed 2023-02-23 by jichenjc