sustainable-computing-io/kepler #570: per core level CPU time accounting in eBPF

rootfs, 2023-03-10:
**Is your feature request related to a problem? Please describe.**
Per conversation with @sunya-ch, early eBPF code has per core level CPU time. This detail helps train and use models with high accuracy. However, on large setup with many cores, the eBPF map may run out of memory, so this detailed CPU time accounting was converted to aggregate CPU time across all the cores. As a result, the model accuracy degraded. 

**Describe the solution you'd like**
The per core CPU time info should be the default format in eBPF. The granularity can change only if the eBPF memory usage is out of control. 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

sunya-ch, 2023-03-10:
Thank you for posting this.
Here is corresponding issue in kepler-model-server project: https://github.com/sustainable-computing-io/kepler-model-server/issues/78

marceloamaral, 2023-03-13:
We don't need that old implementation with memory issues.

To get per-core CPU time, we can just change the key in the eBPF map. There are some hints on how to do that in the code.

The new key will be based on the process PID and CPU ID, so we can group the data by core. But we'll have to make sure how it impacts the code that aggregates the metrics.

sunya-ch, 2023-03-27:
> We don't need that old implementation with memory issues.
>
> To get per-core CPU time, we can just change the key in the eBPF map. There are some hints on how to do that in the code.
>
> The new key will [...]

Thank you, Marcelo. I just check the bpf code according to your point. If I understand correctly, we can just add cpu_id to process_metrics_t 
https://github.com/sustainable-computing-io/kepler/blob/7942754e9324f675ca99572e45a0db8dcb1a5ff0/pkg/bpfassets/perf_event_bindata.go#L91-L101

and update 
https://github.com/sustainable-computing-io/kepler/blob/7942754e9324f675ca99572e45a0db8dcb1a5ff0/pkg/collector/container_hc_collector.go#L38-L48

right?

sunya-ch, 2023-03-27:
> We don't need that old implementation with memory issues.
>
> To get per-core CPU time, we can just change the key in the eBPF map. There are some hints on how to do that in the code.
>
> The new key will [...]

@marceloamaral Thank you. I just checked the bpf code as you pointed out. I'm also afraid of changing the process hash key to (pid, cpu_id) too as it can increase the number of entry and size of hash a lot especially when the number of cpu is large. 

I'm thinking to expose the already-existing pid_time hash to at least giving a mapping which CPUs are assigned to the pod (used by process); even though we don't know the absolute usage value for each core.

marceloamaral, 2023-03-28:
This should be fine, as a process will hardly use all CPUs.

Also, keeping the data in two maps do not solve the hash size problem.

rootfs, 2023-05-10:
05/09 meeting: 
- needs to validate first
- benefit works with fine tune frequency at per core level. 
- VM not to benefit from per core granularity. (VM usually on the same socket)
- large setup has a bpf memory issue
- per socket level tracking
- cpu time per core requires topology level info, passed by user code into ebpf, so ebpf to aggregate per socket. 
- big changes expected to get both cpu time/instruction

sunya-ch, 2023-05-18:
I am now working on this matter and will push a PR soon.
I confirmed that we can collect process metrics on package level with a few changes on BPF code without per-core needed. 

However, instead of export per-package metrics to prometheus which will pretty disrupt current metric collection structure, we can first export new metric as a CPU usage ratio computed from CPU time per package. For training or ratio power modeling, we can use this information separately for CPU-related metric.

The new prometheus metric will look like the following

name: kepler_container_cpu_usage_per_package_ratio (gauge)
value: ratio number
label: container info (pod, container name, namespace), package 

sum(kepler_container_cpu_usage_per_package_ratio) by (container info) will be always equal to 1

marceloamaral, 2023-05-18:
Thanks @sunya-ch for taking care of this!

But can we use a `socket` label in the current metrics?

Including a new label might not impact the existing promQL if the already use the `sum()` function.

closed 2023-08-28 by stale[bot]