sustainable-computing-io/kepler #575: RAPL reset energy value

LeiZhou-97, 2023-03-13:
**Describe the bug**
From the source code, it only sets the Delta when newAggr larger than oldAggr. But RAPL will reset the energy based on the max_energy_range_uj file.

So kepler will treat the reset value as noise. 
https://github.com/sustainable-computing-io/kepler/blob/1e63aa70d458ee66d1460ab4593602a6d9078286/pkg/collector/metric/stats.go#L115

I don't know if it's designed that way, or just a bug.

**Expected behavior**
Not threat this kind of value as noise.

rootfs, 2023-03-13:
The RAPL data reading is getting better on newer platforms.

LeiZhou-97, 2023-03-14:
On platforms with high core counts like 128 cores, the RAPL counter resets about every 10 minutes.

rootfs, 2023-03-14:
@LeiZhou-97 do you use RAPL for other use? Kepler keeps accrued pkg and core joules that are not affected by RAPL reset.

marceloamaral, 2023-04-07:
@LeiZhou-97 do you have any recommendation or we can close this issue?

jiere, 2023-04-07:
@marceloamaral and @rootfs , @LeiZhou-97 is in vacation, yes we can close this issue.

jiere, 2023-04-07:
> @LeiZhou-97 do you use RAPL for other use? Kepler keeps accrued pkg and core joules that are not affected by RAPL reset.

One possible requirement is: do we Kepler support expose a short period of accrued metrics, and if the answer is yes, what's the shortest interval on that?

rootfs, 2023-04-07:
@jiere what about a special log message or metrics for every reset?

jiere, 2023-04-10:
> @jiere what about a special [alert](https://prometheus.io/docs/alerting/latest/alertmanager/) or metrics for every reset?

@rootfs , I personally agree with @marceloamaral 's earlier comments, it seems that such reset is not a deserve-fix issue. And if we raise alert or report specific metrics for it, customers/users may get a little bit confused.

marceloamaral, 2023-04-10:
> One possible requirement is: do we Kepler support expose a short period of accrued metrics, and if the answer is yes, what's the shortest interval on that?

Right now, the interval is hardcoded to 3 second, but we should extend it to have the option to change the pooling interval. Note that RAPL updates usually occur every 100 milliseconds, whereas ACPI updates happen every second. Thus, for intervals shorter than 1 second, we must examine the synchronization logic of the measurements from different power sources.

closed 2023-06-16 by stale[bot]