sustainable-computing-io/kepler #576: Running on AWS VM -- modprobe: FATAL: Module kheaders not found in directory /lib/modules/5.15.0-1031-aws

chateauxvt, 2023-03-13:
**Describe the bug**
The logs of the kepler-exporter are given below. The pod is Running but not functioning properly. 
See the steps to reproduce for details. However, the kconfig.h file is in fact there in the /include/linux folder within the linux header.

Failed to init nvml: could not init nvml: <nil>, using dummy source to obtain gpu power
I0313 16:30:11.169173       1 gpu_nvml.go:45] could not init nvml: <nil>
I0313 16:30:11.170197       1 exporter.go:150] Kepler running on version: 46d55d9
I0313 16:30:11.170257       1 config.go:153] using gCgroup ID in the BPF program: true
I0313 16:30:11.170309       1 config.go:154] kernel version: 5.15
I0313 16:30:11.170349       1 config.go:172] EnabledGPU: true
I0313 16:30:11.170510       1 rapl_msr_util.go:143] failed to open path /dev/cpu/1/msr: no such file or directory
I0313 16:30:11.170600       1 power.go:64] Not able to obtain power, use estimate method
I0313 16:30:12.488995       1 exporter.go:168] Initializing the GPU collector
modprobe: FATAL: Module kheaders not found in directory /lib/modules/5.15.0-1031-aws
<built-in>:1:10: fatal error: './include/linux/kconfig.h' file not found
#include "./include/linux/kconfig.h"
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
I0313 16:30:12.909494       1 bcc_attacher.go:64] failed to attach the bpf program: <nil>
I0313 16:30:12.909528       1 bcc_attacher.go:124] failed to attach perf module with options [-DNUM_CPUS=2 -DSET_GROUP_ID]: failed to attach the bpf program: <nil>, not able to load eBPF modules
I0313 16:30:12.909581       1 exporter.go:185] failed to start : failed to attach bpf assets: failed to attach the bpf program: <nil>
I0313 16:30:12.909756       1 exporter.go:210] Started Kepler in 1.739579451s

**To Reproduce**
We have a single-node K8s cluster set up using RKE v1.4.3. Kubernetes version is v1.24.8 with Docker 20.10.23. The node is an AWS VM with OS Ubuntu 22.04. Some additional debugging was required to successfully launch the pod. The steps are given below:

1) Change "DIrectory" --> "DirectoryOrCreate" for the hostPath volume "/lib/modules" otherwise it could not find the folder.
2) The "build" directory inside /lib/modules/5.15.0-1031-aws is a symlink to "/usr/src/linux-headers-5.15.0-1031-aws" so that also needs to be mounted into the pod. Therefore added a new volumeMount
        - mountPath: /usr/src/linux-headers-5.15.0-1031-aws
          name: lhd
          readOnly: true
with the associated volume
      - hostPath:
          path: /usr/src/linux-headers-5.15.0-1031-aws
          type: DirectoryOrCreate
        name: lhd

**Expected behavior**
For the kepler-exporter to be functioning as it is supposed to.

**Desktop (please complete the following information):**
 - Running on AWS (kernel headers could be modified)
 - OS: Ubuntu 22.04
 -  Kubernetes v1.24.8
 - Docker 20.10.23
 - cgroupsv2 enabled

rootfs, 2023-03-13:
can you try this volume config? We have similar issue testing kepler on github runner, the solution is bind mount the entire `/usr/src` to kepler pod

https://github.com/sustainable-computing-io/kepler/blob/main/manifests/config/exporter/patch/patch-ci.yaml

rootfs, 2023-03-13:
If the VM is started with [host cpu passthrough](https://qemu.readthedocs.io/en/latest/system/qemu-cpu-models.html#two-ways-to-configure-cpu-models-with-qemu-kvm), Kepler can see the HW perf counters.

chateauxvt, 2023-03-19:
Thank you. I believe there is some configuration I'm missing. Closing this issue as the original problem was addressed.

closed 2023-03-19 by chateauxvt