sustainable-computing-io/kepler #577: Docs are missing explanation for kepler_container_dram_joules_total

BGrasnick, 2023-03-13:
On this site:

https://sustainable-computing.io/design/metrics/#kepler-metrics-for-node-energy-consumption

under "Kepler metrics for Node energy consumption"

it states:

"kepler_node_dram_joules_total (Counter) Similar to container metrics, but representing the aggregation of all containers running on the node and operating system (i.e. "system_process")."

but the container metrics explanation for "kepler_container_dram_joules_total" is missing on the page.

rootfs, 2023-03-13:
Good catch. Can you update the section [here](https://github.com/sustainable-computing-io/kepler-doc/blob/main/docs/design/metrics.md#kepler-metrics-for-container-energy-consumption)? Thanks

rootfs, 2023-03-14:
`kepler_container_dram_joules_total` is a metric that describes the total energy spent in DRAM by a container.

marceloamaral, 2023-05-18:
Created a PR to fix this
https://github.com/sustainable-computing-io/kepler-doc/pull/44

closed 2023-05-18 by rootfs