sustainable-computing-io/kepler #586: s390x: failed to build container

jichenjc, 2023-03-15:
```
[root@gustily1 kepler]# make build_containerized
GO111MODULE=off go get -d github.com/go-bindata/go-bindata/...
./hack/bindata.sh
go mod tidy
go mod vendor
./automation/presubmit-tests/gofmt.sh
Checking go format
echo BIN_TIMESTAMP==2023-03-15T10:04:21Z
BIN_TIMESTAMP==2023-03-15T10:04:21Z
/usr/bin/podman build -t quay.io/sustainable_computing_io/kepler:v0.4.10-linux-s390x \
        -f "/root/kepler"/build/Dockerfile \
        --build-arg SOURCE_GIT_TAG=v0.4.10 \
        --build-arg BIN_TIMESTAMP=2023-03-15T10:04:21Z \
        --platform="linux/s390x" \
        .
[1/2] STEP 1/11: FROM quay.io/sustainable_computing_io/kepler_builder:ubi-8.6-bcc-0.24-go1.18 AS builder
Trying to pull quay.io/sustainable_computing_io/kepler_builder:ubi-8.6-bcc-0.24-go1.18...
Getting image source signatures
Copying blob d8b38d01ad98 skipped: already exists
Copying blob 75abf9e8bd60 skipped: already exists
Copying blob 92d1494c70bb skipped: already exists
Copying blob 953dd1512a72 skipped: already exists
Copying blob 04a9758c7322 skipped: already exists
Copying blob 649e58dac065 skipped: already exists
Copying blob bb724c460796 skipped: already exists
Copying blob e759802f8f09 skipped: already exists
Copying config 1ceb9c1220 done
Writing manifest to image destination
Storing signatures
WARNING: image platform ({amd64 linux  [] }) does not match the expected platform ({s390x linux  [] })
[1/2] STEP 2/11: ARG BIN_TIMESTAMP
--> Using cache 09183d36a66e04d8c256e1914435da21206aa9d4c26f7e97714807b56afc650b
--> 09183d36a66
[1/2] STEP 3/11: ARG SOURCE_GIT_TAG
--> Using cache 23f1e9448733abcec4f8191939fa29bc1bdec1a93f92756966e4b276cbe61583
--> 23f1e944873
[1/2] STEP 4/11: LABEL name=kepler-builder
--> Using cache 0995ab56a3e75f179cf84a8826e6afafdc7f514329821241fe4fdd5c7693b54d
--> 0995ab56a3e
[1/2] STEP 5/11: ENV GOPATH=/opt/app-root GO111MODULE=on GOROOT=/usr/local/go
--> Using cache 158983cb3b852e2902d9a2dc5504d819086d15360e2643346109f02f972bfd61
--> 158983cb3b8
[1/2] STEP 6/11: ENV PATH=$GOPATH/bin:$GOROOT/bin:$PATH
--> Using cache d83620a10ed74027f904d563017c35a4a3c18c895bfc2ff8f2b01ee832fb3987
--> d83620a10ed
[1/2] STEP 7/11: WORKDIR $GOPATH/src/github.com/sustainable-computing-io/kepler
--> Using cache 5dff276aa74edad98e802c4b9cd530a515eebf781e2c7a8c8733c550baca4be4
--> 5dff276aa74
[1/2] STEP 8/11: COPY . .
--> 98326adab50
[1/2] STEP 9/11: RUN mkdir -p data
exec container process `/bin/sh`: Exec format error
Error: error building at STEP "RUN mkdir -p data": error while running runtime: exit status 1
make: *** [Makefile:108: build_containerized] Error 1

```

@SamYuan1990 is it supposed to be working? is it because there is no kepler build s390x version?

SamYuan1990, 2023-03-15:
> ```
> [root@gustily1 kepler]# make build_containerized
> GO111MODULE=off go get -d github.com/go-bindata/go-bindata/...
> ./hack/bindata.sh
> go mod tidy
> go mod vendor
> ./automation/presubmit-tests/gofmt.sh [...]

@rootfs do we have s390x base image right?
I suppose the error is deal to `sh` on amd64 is different with s390x as format error.

rootfs, 2023-03-15:
the builder image doesn't have multi arch yet...

SamYuan1990, 2023-03-15:
ok... as we have merged pr for image builder with buildx, I will take time on base image with buildx and multi arch the base image.

closed 2023-05-24 by stale[bot]