sustainable-computing-io/kepler #591: RAPL code issues

jiere, 2023-03-17:
**Describe the bug**
There are some code issues on pkg/power module RAPL side.
1. In pkg/power/component/power.go, for platforms which support RAPL sysfs, there is no chance to reach the RAPL msr side, not sure if the rapl_msr*.go are dead code or not?
2. In pkg/power/component/source/rapl_sysfs_util.go, current code to get cpuNum and cpuPackage is low efficient, could be refined.
3. In pkg/power/component/source/rapl_msr_util.go, current code for CPU package and core mapping has logical error.
See function "mapPackageAndCore" in rapl_msr_util.go:
https://github.com/sustainable-computing-io/kepler/blob/main/pkg/power/components/source/rapl_msr_util.go#L81
The "id" here is the specific core's physical_package_id, i.e. On two-sockets platforms, the "id" should be either 0 or 1, so the "packageMap" concept is wrong after the for loop.

**Expected behavior**
1. See function "getNumPackage" and "getNumCPUs" in rapl_sysfs_util.go.
For Server CPUs like Intel 4th Xeon SP and AMD EYPC Genoa, etc, there might be hundreds of cores per package, the more efficient way to get cpu number and package number is leveraging the output of "lscpu“ command:
Ubuntu system example:
$ lscpu |grep -E "CPU\(s\)|Socket\(s\)"
CPU(s):                          128
On-line CPU(s) list:             0-127
Socket(s):                       2
NUMA node0 CPU(s):               0-31,64-95
NUMA node1 CPU(s):               32-63,96-127

RHEL system example:
$ lscpu |grep -E "CPU\(s\)|Socket\(s\)"
CPU(s):              224
On-line CPU(s) list: 0-223
Socket(s):           2
NUMA node0 CPU(s):   0-55,112-167
NUMA node1 CPU(s):   56-111,168-223

For CPU number, maybe need some more code to fetch the exact number; for package number, just "lscpu |grep Socket" is enough.

2. If the RAPL MSR code is not dead code, we need to fix the issue mentioned above.

jiere, 2023-03-17:
> @jiere On certain environment (hyper V if my memory is right), the RAPL sysfs files are missing, but msr are accessible. That gave rise to the msr code. If you find that is not the case any more, we can remove the msr code.

Thanks @rootfs for your quick reply, I will check later and get you tuned.

jichenjc, 2023-03-20:
my hardware (x86 PC) support msr but no RAPL_sysfs, so the MSR is used in my env..

```
[root@localhost ~]# ls /dev/cpu/0/msr
/dev/cpu/0/msr
[root@localhost ~]# ls /sys/class/powercap/intel-rapl/
ls: cannot access /sys/class/powercap/intel-rapl/: No such file or directory
[root@localhost ~]#
```

jiere, 2023-03-21:
Update from my side, I tried on my laptop:
host CPU: 11th Gen Intel(R) Core(TM) i7-1185G7 @ 3.00GHz
host OS: Windows 11 Enterprise

I enabled Hyper-V and provisioned a new VM:
jie@jie-Virtual-Machine:~/Desktop$ uname -r
5.15.0-27-generic
jie@jie-Virtual-Machine:~/Desktop$ cat /etc/os-release
PRETTY_NAME="Ubuntu 22.04 LTS"
...

In the VM's sysfs, there is dTPM framework visible under powercap, but no intel-rapl also.

jie@jie-Virtual-Machine:~/Desktop$ ls /sys/class/powercap
dtpm
jie@jie-Virtual-Machine:~/Desktop$ ls /sys/class/powercap/dtpm -l
lrwxrwxrwx 1 root root 0 Mar 20  2023 /sys/class/powercap/dtpm -> ../../devices/virtual/powercap/dtpm

While msr path is accessible: 
jie@jie-Virtual-Machine:~/Desktop$ ls /dev/cpu/0/msr
/dev/cpu/0/msr

jiere, 2023-03-21:
> my hardware (x86 PC) support msr but no RAPL_sysfs, so the MSR is used in my env..
> 
> ```
> [root@localhost ~]# ls /dev/cpu/0/msr
> /dev/cpu/0/msr
> [root@localhost ~]# ls /sys/class/powercap/intel-rapl/
> ls: cannot access /sys/class/powercap/intel-rapl/: No such file or directory
> [root@localhost ~]#
> ```

Could you also list your PC CPU and OS info? Thanks.

jichenjc, 2023-03-21:
here you go
```
[root@localhost ~]# lscpu
Architecture:          x86_64
CPU op-mode(s):        32-bit, 64-bit
Byte Order:            Little Endian
CPU(s):                32
On-line CPU(s) list:   0-31
Thread(s) per core:    2
Core(s) per socket:    8
Socket(s):             2
NUMA node(s):          2
Vendor ID:             GenuineIntel
CPU family:            6
Model:                 45
Model name:            Intel(R) Xeon(R) CPU E5-2470 0 @ 2.30GHz
Stepping:              7
CPU MHz:               1199.902
CPU max MHz:           3100.0000
CPU min MHz:           1200.0000
BogoMIPS:              4599.59
Virtualization:        VT-x
L1d cache:             32K
L1i cache:             32K
L2 cache:              256K
L3 cache:              20480K
NUMA node0 CPU(s):     0-7,16-23
NUMA node1 CPU(s):     8-15,24-31
Flags:                 fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca                                                                                                              cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1g                                                                                                             b rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_t                                                                                                             sc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx                                                                                                             16 xtpr pdcm pcid dca sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer aes xsave a                                                                                                             vx lahf_lm epb pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid xs                                                                                                             aveopt dtherm ida arat pln pts md_clear flush_l1d
[root@localhost ~]# cat /etc/os-release
NAME="CentOS Linux"
VERSION="7 (Core)"
ID="centos"
ID_LIKE="rhel fedora"
VERSION_ID="7"
PRETTY_NAME="CentOS Linux 7 (Core)"
ANSI_COLOR="0;31"
CPE_NAME="cpe:/o:centos:centos:7"
HOME_URL="https://www.centos.org/"
BUG_REPORT_URL="https://bugs.centos.org/"

CENTOS_MANTISBT_PROJECT="CentOS-7"
CENTOS_MANTISBT_PROJECT_VERSION="7"
REDHAT_SUPPORT_PRODUCT="centos"
REDHAT_SUPPORT_PRODUCT_VERSION="7"

[root@localhost ~]# uname -a
Linux localhost.localdomain 5.4.231-1.el7.elrepo.x86_64 #1 SMP Sat Feb 4 13:55:47 EST 2023 x86_64 x86_64 x86_64 GNU/Linux

```

jiere, 2023-03-23:
@jichenjc , please try below command to check if the RAPL is not enabled in your current kernel config:
$ grep RAPL /boot/config-$(uname -r)
CONFIG_PERF_EVENTS_INTEL_RAPL=y
CONFIG_PROC_THERMAL_MMIO_RAPL=m
CONFIG_INTEL_RAPL_CORE=m
CONFIG_INTEL_RAPL=m

jichenjc, 2023-03-23:
um.. you are right 

my env didn't enable RAPL currently seems

```
# grep RAPL /boot/config-$(uname -r)
CONFIG_PERF_EVENTS_INTEL_RAPL=m
```

but the MSR seems ok to read

jiere, 2023-03-23:
Not sure if the MSR socket is workable on your side, I used current kepler code and wrote a test main.go to check the result, different platforms have different behavior.. some of them are unexpected..
Let me figure out a proper way to fix, thanks for your confirm anyway.

closed 2023-04-07 by jiere