sustainable-computing-io/kepler #593: Cannot access kepler metrics in prometheus

annis-souames, 2023-03-19:
Hello, 
I am trying to install Kepler in a Kubernetes cluster on Google cloud (GKE) to monitor energy consumption. Afterwards, I installed prometheus and grafana using helm. However I can't find `kepler_` metrics in prometheus metric explorer ? 

Are there any other steps to take to get Kepler metrics in Prometheus ?

jichenjc, 2023-03-20:
Please try whether you can see the output from `curl` command? 
see the demo https://www.youtube.com/watch?v=P5weULiBl60

annis-souames, 2023-03-20:
The curl command gives the metrics correctly, I can also access the metrics through a web browser by port-forwarding kepler-exporter. But they are not visible in Prometheus. I believe there is no service Monitor running for Kepler and for Prometheus since I installed it with helm from a different source, I will try to re-install prometheus following the documentation.

Note : When I run, `kubectl get servicemonitor` gives `No resource found`.

rootfs, 2023-04-07:
@annis-souames is it still an issue?

annis-souames, 2023-04-07:
Yes the issue is solved, thank you

closed 2023-04-07 by annis-souames