sustainable-computing-io/kepler #594: Cannot see the metrics of pods created after Kepler

andersonandrei, 2023-03-20:
**Describe the bug**

I'm running a serverless platform on top of AK8, so, for each function that I execute in the platform, a new pod is created. Besides, Kepler does not export metrics for those pods new pods. It just export metrics for the pods that were already in the platform before its (Kepler) own deployment. 
For instance, let us consider the following actions: 1) to deploy kepler, 2) to create a new pod, and 3) to delete kepler and deploy it again. If I do 1) and 2), Kepler does not export such pod metrics, so I need to do 1), 2) and 3), and still, the problem persists with new pods created after 3).

I’m using the Kepler interface through Prometheus, but I try the queries with the Kepler metric endpoint with curl from time to time as well. In both cases, I can just see new pods’ metrics after steps 1), 2), and 3).

**To Reproduce**
Steps to reproduce the behavior:

Deployed Kepler manually, using one of the manifest available, with slight modifications:

```
kind: Namespace
metadata:
  labels:
    sustainable-computing.io/app: kepler
  name: kepler
---
apiVersion: v1
kind: ServiceAccount
metadata:
  labels:
    sustainable-computing.io/app: kepler
  name: kepler-sa
  namespace: monitoring
---
apiVersion: rbac.authorization.k8s.io/v1
kind: Role
metadata:
  labels:
    app.kubernetes.io/component: prometheus
    app.kubernetes.io/instance: k8s
    app.kubernetes.io/name: prometheus
    sustainable-computing.io/app: kepler
  name: prometheus-k8s
  namespace: monitoring
rules:
- apiGroups:
  - ""
  resources:
  - services
  - endpoints
  - pods
  verbs:
  - get
  - list
  - watch
- apiGroups:
  - extensions
  resources:
  - ingresses
  verbs:
  - get
  - list
  - watch
- apiGroups:
  - networking.k8s.io
  resources:
  - ingresses
  verbs:
  - get
  - list
  - watch
---
apiVersion: rbac.authorization.k8s.io/v1
kind: ClusterRole
metadata:
  labels:
    sustainable-computing.io/app: kepler
[... 499 lines cut ...]
    lastTransitionTime: "2023-03-03T14:34:00Z"
    status: "True"
    type: PodScheduled
  containerStatuses:
  - containerID: containerd://fa495a86a17d5ca721517533fbbb5659d718f942d61ffbbd383a91befbf3a6de
    image: quay.io/sustainable_computing_io/kepler:release-0.4
    imageID: quay.io/sustainable_computing_io/kepler@sha256:67c34e1ade5f17cc444aa134f7d95b424077af6bc7c05d2ff82d536a3e0a6174
    lastState: {}
    name: kepler-exporter
    ready: true
    restartCount: 0
    started: true
    state:
      running:
        startedAt: "2023-03-03T14:34:02Z"
  hostIP: 10.224.0.5
  phase: Running
  podIP: 10.244.1.89
  podIPs:
  - ip: 10.244.1.89
  qosClass: Burstable
  startTime: "2023-03-03T14:34:00Z"
```

Thanks a lot!

rootfs, 2023-03-20:
I don't have an AKS setup but let me try it on my k8s setup and get back to you. I am bit tied up, response delay is all possible.

rootfs, 2023-03-20:
i ran the pod on my k8s node, the pod metrics are there
[image]

this is the yaml i used
```yaml
apiVersion: v1
kind: Pod
metadata:
  name: wskowdev-invoker-00-43-guest-linpack
  namespace: monitoring
spec:
  containers:
  - env:
    - name: __OW_API_HOST
      value: https://ourserver.io:31001
    - name: __OW_ALLOW_CONCURRENT
      value: "false"
    image: andersonandrei/python3action:linpack
    imagePullPolicy: IfNotPresent
    name: user-action
    ports:
    - containerPort: 8080
      name: action
      protocol: TCP
    resources:
      limits:
        memory: 256Mi
      requests:
        memory: 256Mi
    securityContext:
      capabilities:
        drop:
        - NET_RAW
        - NET_ADMIN
       ```

andersonandrei, 2023-03-20:
@rootfs , thanks for your answer.

Have you deployed Kepler before or after deploying the linpack pod? I don't see it in the list of pods you showed above. 
Can you tell me how you deployed Kepler, please?

rootfs, 2023-03-20:
@andersonandrei i sent you a slack invite, let's sync up there and do live debugging

rootfs, 2023-04-07:
I am able to reproduce this issue on a nested VM environment (but no on bare metal machines). 

I think the root cause is that kubelet and kepler have a split brain in understanding the newly created containers. When kepler receives the container ID and resolves it through Kubelet, Kubelet has not updated the container with Pod name and namespace yet. So kepler just uses `system_process` as the Pod name and saves the container ID in the cache. As a result, the container fails to show up in the kepler metrics.

To fix this, we can try the following:
- delay the container ID resolution, if this container has a crio sysfs path, so kepler have chance to retry querying kubelet until timeout.  This can be prototyped quickly.
- expose the container ID as it is and wait for external service to resolve it by querying kube-apiserver (#301)

cc @marceloamaral

marceloamaral, 2023-04-10:
@rootfs we indeed have this problem and I am currently working on the kube-apiserver.

@andersonandrei could you also share your Kepler logs?

rootfs, 2023-04-10:
@marceloamaral that's cool! I think the higher level APIs are more useful than Pod for end users. In that context, querying kube-apiserver in a new service will also solve this problem nicely.

andersonandrei, 2023-04-11:
> 

Hi @marceloamaral , thanks for your help :)

Here are the logs:

`kubectl logs kepler-exporter-bb498 -n openwhisk
I0411 09:34:40.060737       1 gpu.go:43] Failed to init nvml, err: could not init nvml: error opening libnvidia-ml.so.1: libnvidia-ml.so.1: cannot open shared object file: No such file or directory
I0411 09:34:40.063029       1 exporter.go:149] Kepler running on version: 5b9c91a
I0411 09:34:40.063065       1 config.go:169] using gCgroup ID in the BPF program: true
I0411 09:34:40.063093       1 config.go:170] kernel version: 4.19
I0411 09:34:40.063170       1 power.go:53] use sysfs to obtain power
I0411 09:34:40.371612       1 exporter.go:167] Initializing the GPU collector
I0411 09:34:41.378660       1 bcc_attacher.go:135] Successfully load eBPF module with option: [-DNUM_CPUS=32]
I0411 09:34:41.395018       1 exporter.go:209] Started Kepler in 1.33200242s`

jichenjc, 2023-04-14:
>>>So kepler just uses system_process as the Pod name and saves the container ID in the cache. As a result, the container fails to show up in the kepler metri

@rootfs if we put into system_process then we will NEVER put them back even though kubelet detect such? or it's just a small time window issue?

>>>expose the container ID as it is and wait for external service to resolve it by querying kube-apiserve

we said we want to avoid kube-api  access due to performance issue, will this be a problem?

jichenjc, 2023-04-17:
>>>The kube-apiserver query is going to be done in a separate exporter that takes the container ID, Pod and Namespace meta data, to aggregate to Deployment/Daemonset/StatefulSet

ok, you mean an exporter other than kepler exporter? anyway, code update will be helpful to know what's really proposed here

closed 2023-08-28 by stale[bot]