sustainable-computing-io/kepler #595: How to build manifests manually ?

annis-souames, 2023-03-20:
I've been trying to build manifests manually for Kepler for a few hours with no luck. I don't understand how to run `build-manifest.sh` ? 
There is no build-manifest in the root of the project, while the Makefile is in the root of the project. Should I move build-manifest.sh from hack folder to the root of Kepler folder ?  I am a beginner with `make`. 

The documentation is a bit broad about this, could you please explain how I can build the manifests manually ? I suspect that the helm installation is not working correctly, and I want to install Kepler manually. 

If I move build-manifest.sh from hack folder and then run it through `make build-manifest.sh` , I get the following output : 

```
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
make: command: Command not found
make: command: Command not found
mkdir -p bin
go: creating new go.mod: module tmp
Downloading sigs.k8s.io/kustomize/kustomize/v4@v4.5.2
can't load package: package sigs.k8s.io/kustomize/kustomize/v4@v4.5.2: can only use path@version syntax with 'go get'
make: *** [Makefile:241: kustomize] Error 1
```

rootfs, 2023-03-20:
please go to the `Build manifests` section here https://sustainable-computing.io/installation/kepler/

annis-souames, 2023-03-20:
I followed that section but there is no build-manifest file in the root of kepler folder, they are all in hack folder, which folder I should run the `make` command in ?

rootfs, 2023-03-20:
In the top level of kepler (it does require go and kubectl though).

[image]

Or you can try the operator or helm chart deployments

annis-souames, 2023-03-20:
> @annis-souames I can build a manifests for you to try. What options do you need for the manifest, e.g. bare metal or VM? If on VMs, have you installed kernel headers on your env?

That would be so nice, I am using Google Kubernetes Engine so yes it's a VM, basically I am trying to monitor the energy consumption of this cluster on GKE and get the results on prometheus. 

> have you installed kernel headers on your env?

Yes, I updated the nodes of the cluster to use cgroups v2

rootfs, 2023-03-20:
```yaml
apiVersion: v1
kind: Namespace
metadata:
  labels:
    sustainable-computing.io/app: kepler
  name: kepler
---
apiVersion: v1
kind: ServiceAccount
metadata:
  labels:
    sustainable-computing.io/app: kepler
  name: kepler-sa
  namespace: kepler
---
apiVersion: rbac.authorization.k8s.io/v1
kind: ClusterRole
metadata:
  labels:
    sustainable-computing.io/app: kepler
  name: kepler-clusterrole
rules:
- apiGroups:
  - ""
  resources:
  - nodes/metrics
  - nodes/proxy
  - nodes/stats
  verbs:
  - get
  - watch
  - list
---
apiVersion: rbac.authorization.k8s.io/v1
kind: ClusterRoleBinding
metadata:
  labels:
    sustainable-computing.io/app: kepler
  name: kepler-clusterrole-binding
roleRef:
  apiGroup: rbac.authorization.k8s.io
  kind: ClusterRole
  name: kepler-clusterrole
subjects:
- kind: ServiceAccount
  name: kepler-sa
  namespace: kepler
---
apiVersion: v1
data:
  BIND_ADDRESS: 0.0.0.0:9102
  CGROUP_METRICS: '*'
  CPU_ARCH_OVERRIDE: ""
  ENABLE_EBPF_CGROUPID: "true"
  ENABLE_GPU: "true"
  ENABLE_PROCESS_METRICS: "false"
  EXPOSE_IRQ_COUNTER_METRICS: "true"
  EXPOSE_KUBELET_METRICS: "true"
  KEPLER_LOG_LEVEL: "1"
  KEPLER_NAMESPACE: kepler
  METRIC_PATH: /metrics
  MODEL_CONFIG: |
    CONTAINER_COMPONENTS_ESTIMATOR=false
    CONTAINER_COMPONENTS_INIT_URL=https://raw.githubusercontent.com/sustainable-computing-io/kepler-model-server/main/tests/test_models/DynComponentModelWeight/CgroupOnly/ScikitMixed/ScikitMixed.json
kind: ConfigMap
metadata:
  labels:
    sustainable-computing.io/app: kepler
  name: kepler-cfm
  namespace: kepler
---
apiVersion: v1
kind: Service
metadata:
  labels:
    app.kubernetes.io/component: exporter
    app.kubernetes.io/name: kepler-exporter
    sustainable-computing.io/app: kepler
  name: kepler-exporter
  namespace: kepler
spec:
  clusterIP: None
  ports:
  - name: http
    port: 9102
    targetPort: http
  selector:
    app.kubernetes.io/component: exporter
    app.kubernetes.io/name: kepler-exporter
    sustainable-computing.io/app: kepler
---
apiVersion: apps/v1
kind: DaemonSet
metadata:
  labels:
    sustainable-computing.io/app: kepler
  name: kepler-exporter
  namespace: kepler
spec:
  selector:
    matchLabels:
      app.kubernetes.io/component: exporter
      app.kubernetes.io/name: kepler-exporter
      sustainable-computing.io/app: kepler
  template:
    metadata:
      labels:
        app.kubernetes.io/component: exporter
        app.kubernetes.io/name: kepler-exporter
        sustainable-computing.io/app: kepler
    spec:
      containers:
      - args:
        - /usr/bin/kepler -v=1
        command:
        - /bin/sh
        - -c
        env:
        - name: NODE_NAME
          valueFrom:
            fieldRef:
              fieldPath: spec.nodeName
        image: quay.io/sustainable_computing_io/kepler:latest
        imagePullPolicy: Always
        livenessProbe:
          failureThreshold: 5
          httpGet:
            path: /healthz
            port: 9102
            scheme: HTTP
          initialDelaySeconds: 10
          periodSeconds: 60
          successThreshold: 1
          timeoutSeconds: 10
        name: kepler-exporter
        ports:
        - containerPort: 9102
          name: http
        resources:
          requests:
            cpu: 100m
            memory: 400Mi
        securityContext:
          privileged: true
        volumeMounts:
        - mountPath: /usr/src
          name: usr-src
        - mountPath: /lib/modules
          name: lib-modules
        - mountPath: /sys
          name: tracing
        - mountPath: /proc
          name: proc
        - mountPath: /etc/kepler/kepler.config
          name: cfm
          readOnly: true
      dnsPolicy: ClusterFirstWithHostNet
      serviceAccountName: kepler-sa
      tolerations:
      - effect: NoSchedule
        key: node-role.kubernetes.io/master
      volumes:
      - hostPath:
          path: /usr/src
          type: Directory
        name: usr-src
      - hostPath:
          path: /proc-host
          type: Directory
        name: proc
      - hostPath:
          path: /lib/modules
          type: Directory
        name: lib-modules
      - hostPath:
          path: /sys
          type: Directory
        name: tracing
      - configMap:
          name: kepler-cfm
        name: cfm
```

annis-souames, 2023-03-20:
Thank you so much, you just saved me from giving up on a project I've been working on for a month ! I will try to deploy these right away.

Thank's again !

rootfs, 2023-03-20:
@annis-souames looking forward to your good progress!

closed 2023-04-07 by rootfs