sustainable-computing-io/kepler #598: Resolve containers

rinana, 2023-03-24:
Hi
I'm trying to install Kepler on an openshift cluster and an error occurs. 
The log of the kepler-exporter pod is as follows: 

```I0323 15:55:54.319760       1 bcc_attacher.go:135] Successfully load eBPF module with option: [-DNUM_CPUS=4 -DSET_GROUP_ID]
I0323 15:55:54.326436       1 resolve_container.go:144] failed to get response: failed to get response from "https://XXXXXXXX:10250/pods": Get "https://XXXXXXXX:10250/pods": dial tcp: lookup XXXXXXXX on Y.Y.Y.Y:53: no such host
```

[podLister.ListPods() ](https://github.com/sustainable-computing-io/kepler/blob/main/pkg/kubelet/kubelet_pod_lister.go#L95) returns this error...
How do I resolve it?

sunya-ch, 2023-03-24:
XXXXXXXX refers to node name. 

The host cannot resolve the IP of the node name. 
I wonder why can't we just use the localhost to connect to kubelet.
Do we expect remote kubelet? In that case, I think we might rename the environment name. 

btw, we can also get hostname from `os.Hostname()`.

https://github.com/sustainable-computing-io/kepler/blob/7942754e9324f675ca99572e45a0db8dcb1a5ff0/pkg/kubelet/kubelet_pod_lister.go#L56-L67

jichenjc, 2023-03-24:
>>>I wonder why can't we just use the localhost to connect to kubelet.

localhost will be the pod that running kepler ,correct? pod and node has different tcpip stack so I am wondering localhost will not be the right endpoint

>>>May try removing the environment NODE_NAME from the deployment manifest.

not sure whether we need support https://virtual-kubelet.io/ (I guess not) but this might need further discussion

rinana, 2023-03-24:
In my environment, using localhost instead of node name does not work...
I'm not sure it's a configuration of openshift or just our environment.

but I replaced reference path for the NODE_NAME to `status.hostIP` in the yaml instead of  `spec.nodeName` and it worked.

jichenjc, 2023-03-24:
as I mentioned, the pod running kepler is not able to use localhost to get node info, so that's what you already see

and there must be some issue in your pod DNS setting or it didn't include the hostname of the node into DNS record list
so you can't access through DNS hostname and ip works because it can access.. so it's not kepler unique issue

thus, maybe we can consider provide a config item like IP/HOSTNAME?

rootfs, 2023-03-27:
@rinana thanks for the solution, can you post a PR?

rinana, 2023-03-28:
@rootfs I posted PR #604

SamYuan1990, 2023-03-28:
may I have two questions here?
- if we specific for hostname/ip address, and combine with a port number there... Hence why not just have an env just as `hostname:port` ?
- should we adjust helm and operator for any environment value changes/updates?

rootfs, 2023-03-28:
closed by #604

rootfs, 2023-03-28:
thank you @rinana

closed 2023-03-28 by rootfs