sustainable-computing-io/kepler #605: Refactor for code seems?

SamYuan1990, 2023-03-28:
https://github.com/sustainable-computing-io/kepler/blob/3f117b3c8b198806a7f26cc0e46bb18830bb0789/pkg/collector/container_cgroup_collector.go#L30-L39

In recently I am checking code blocks above, and here are some thing I request refactor:
Overall speaking, in code blocks above, 
we generate stat Readers for each container, and ... considering that `updateCgroupMetrics` is invoked every 3 seconds, hence as there is 
- [ ] No cache hence each time rebuild for stat reader
Which means for a running container/pod, we rebuild it's statReaders for each time, with out cache.
in my point of view, the statReaders just contains as Path for files, hence for a running container in Cgroup, those path are static, and we don't need rebuild.
- [ ] Data handle couple with fetch data
and then the next item comes, as `func convertToStandard(stats map[string]interface{}) map[string]interface{}` .
which means each time we construct data handler with data.(fetch data with construct data handler at same time)
- [ ] Force type convert 
further more, as we have to use `readVal := cgroupFSValue.(uint64)` as force type convert to set up value.

marceloamaral, 2023-03-28:
@SamYuan1990 good catch!

I do agree that we need to improve this.

Btw, what is the status of the investigation to use an external library to collect the cgroup metrics?
For example, as kublet use cadvisor, can we also use it?
https://github.com/kubernetes/kubernetes/blob/master/pkg/kubelet/cadvisor/cadvisor_linux.go#L36

SamYuan1990, 2023-03-28:
> https://github.com/kubernetes/kubernetes/blob/master/pkg/kubelet/cadvisor/cadvisor_linux.go#L36

em ? https://github.com/sustainable-computing-io/kepler/discussions/558#discussioncomment-5275875
does anyone work on it?

jichenjc, 2023-03-30:
I assume this will be combined with #606（remove kubelet and use cgroup instead) because of we anyway need update code to adopt cgroupv1/v2 ? @marceloamaral @SamYuan1990 
otherwise I can spend some time on it

SamYuan1990, 2023-03-30:
> I assume this will be combined with #606（remove kubelet and use cgroup instead) because of we anyway need update code to adopt cgroupv1/v2 ? @marceloamaral @SamYuan1990 otherwise I can spend some time on #588( switch to containerd/cgroup)

I think the final outcome for this issue is same with #606

jichenjc, 2023-03-30:
then let's see how #606 goes and see what else need to be enhanced

marceloamaral, 2023-03-30:
@jichenjc and @SamYuan1990 I am working on the new impl for cgroup metrics.

closed 2023-04-09 by SamYuan1990